Azure/Industrial-IoT #4: Some questions in doing the OPC twin project

jznling, 2019-06-14:
Hi, I am running OPC twin on a Windows10, Docker 3.7.0, surface pro 4 in China. I work on the OPC twin project following procedures on https://docs.microsoft.com/bs-latn-ba/azure/iot-accelerators/howto-opc-twin-deploy-existing

1. when deploying this project to my Azure portal using Powershell, it kept warning me with "Get-AzureRMxxx is not recognized as a name of cmdlet". This was solved after installing AzureRM module. However, on the web https://docs.microsoft.com/bs-latn-ba/azure/iot-accelerators/howto-opc-twin-deploy-existing, it only requires Az. The Microsoft docs also said that "The Az module is not guaranteed to be compatible with any existing AzureRM installs in PowerShell 5.1 for Windows. ". It is really weird. 

2. In the deploy project part, it requires the user to save the .env file. This requirement is really painful for me. For no obvious reason, the .env file saving command did not show up for half an hour after the Azure resource group was successfully deployed. For each .env file, it took me about 20 to 60 mins to wait for saving command jumped out. In the later deployment, I did not see the importance of these painfully saved .env files. It just showed which resource group I should connect to. Is it really vital to save the .env file? 

3. In the "run and debug locally" part, the deployment to iot edge was not successfully done. The "iotedgehubdev start -d deployment.json" command gets the warning "ERROR: local variable 'module_content' referenced before assignment". All previous steps are correctly followed. Please reply to me if you have any clue to fix the error.

karok2m, 2019-07-10:
Hi @jznling. Thank you for reporting the issues. Let me go over the points one-by-one.

1. You are correct, documentation wrongly mentions `Az` PowerShell module, when it needs to mention `AzureRM`. This is a mistake and will be fixed shortly.

2. `.env` file contains variables required for running the components locally using `docker-compose`, Visual Studio or VS Code. The variables are mostly resource identifiers and connection strings of the resources that are deployed to Azure cloud by `deploy.cmd`. As a result, deployment script has to wait for all the resources to be started, before it can get their connection strings. So saving the `.env` file is quite vital if you want to run or debug components locally.

3. Support for deployment through `iotedgehubdev` has been dropped. Documentation has been updated to provide a deployment using `docker-compose` for local development and through IoT Edge runtime for production use cases. Please check updated documentation for step-by-step guides.

karok2m, 2019-07-26:
Hi @jznling. I've fixed the link in the documentation to point to the correct PowerShell module. So I will close this issue for now. Feel free to reopen it if you think that I've missed anything.

closed 2019-07-26 by karok2m