Azure/Industrial-IoT #9: Gateway endpoint URL issue

tbn-mm, 2019-07-23:
Hi,

I can browse/discover OPC UA servers using UaExpert (an UPA UA Client) and the Gateway service (e.g. http://192.168.135.131:10080/ua/). If an endpoint is activated and connected, I can see 3 different endpoints (http, opc-ua none and sign&encrypt).
The first endpoint provides a localhost URL
`http://localhost:9040/applications/.../endpoint/...`
The second and third endpoint url contains an hostname:
`opc.tcp://b338c9b7c250:51111/applications/.../endpoint/...`

The Client won't be able to connect to these endpoints, because there is no service running on localhost and hostname to IP is not resolvable.
Manually changing localhost to IP is not accepted by the gateway (BadTcpEndpointUrlInvalid), adding `192.168.135.131 b338c9b7c250` to hosts works but bad workaround and doesn't work when using nginx proxy.

cristipogacean, 2019-07-29:
We display (and use) the discovered endpoints as they are reported by the server. Normally it is server's responsibility not to report localhost or not resolvable urls in the get endpoints request.
Nevertheless, there are a few optimization points that can be implemented:
1) http protocol was depricated in opc ua and is not suppoted in our stack, so we should just drop such endpoints - this is the reason your first endpoint is not working even when you use the ip address
2) when hostname is not resolvable, we could potentially replace it with the ip used to discover the endpoint. Nevertheless, the source of the problem is probably a dns infrastructure issue that is independent of our modules.

marcschier, 2021-04-13:
Gateway is deprecated and remove.

closed 2021-04-13 by marcschier