Azure/Industrial-IoT #13: Don't convert NodeId if it has a namespace id

tbn-mm, 2019-07-24:
If you have a node that has the NodeId `ns=2;i=10157` and `NamespaceUris[2] == http://test.org/UA/Data/` then the NodeId changes to `http://test.org/UA/Data/#i=10157`.
If you do `/v2/browse/{endpointId}` with NodeId = `http://test.org/UA/Data/#i=10157` you will get the expected response (some referenced nodes).

There is an OPC UA server that has a `NamespaceUris[4]` like `abc/xyz` without any `http://` prefix and `ns=4;i=123` results in `abc/xyz#i=123`. For some reason the `/v2/browse/{endpointId}` returns the root nodes and not the expected referenced nodes from `ns=4;i=123`.
I compared it with the software UaExpert and this program uses `ns=4;i=123` instead of `abc/xyz#i=123` which results in the expected response.

So I searched the twin modul source code and found a way to remove the conversion to `abc/xyz#i=123` and now I can browse the node as expected.

My request is:
https://github.com/Azure/azure-iiot-opc-ua/blob/e88aaf8b16a74f2016fbf5f6ec64439ec0a6516a/src/Microsoft.Azure.IIoT.OpcUa.Edge/src/Control/AddressSpaceServices.cs#L782 
change to 
`var id = nodeId.ToString();`

cristipogacean, 2019-07-29:
fixed in master

marcschier, 2019-09-07:
In 2.5.1

closed 2019-09-07 by marcschier