Azure/Industrial-IoT #17: endpoint activate causes twin module to restart

tbn-mm, 2019-07-24:
When using `/v2/endpoints/{endpointId}/activate` from Registry Service the twin module shows this message:

```
[13:34:14 ERR] Exception applying changes! Continue...
System.ArgumentNullException: Value cannot be null.
Parameter name: endpoint
   at Microsoft.Azure.IIoT.OpcUa.Protocol.Services.ClientServices.ExecuteServiceAsync[T](EndpointModel endpoint, CredentialModel elevation, Int32 priority, Func`2 service, Nullable`1 timeout, CancellationToken ct, Func`2 handler)
   at Microsoft.Azure.IIoT.OpcUa.Protocol.EndpointServicesEx.ExecuteServiceAsync[T](IEndpointServices client, EndpointModel endpoint, CredentialModel elevation, TimeSpan timeout, Func`2 service)
   at Microsoft.Azure.IIoT.Modules.OpcUa.Twin.v2.Supervisor.NodeSettingsController.WriteAsync(String nodeId, JToken value) in /src/src/v2/Controllers/NodeSettingsController.cs:line 96
   at Microsoft.Azure.IIoT.Modules.OpcUa.Twin.v2.Supervisor.NodeSettingsController.ApplyAsync() in /src/src/v2/Controllers/NodeSettingsController.cs:line 80
   at Microsoft.Azure.IIoT.Module.Framework.Hosting.SettingsRouter.Controller.SafeApplyAsync()
[13:34:14 DBG] Applying new settings took 00:00:00.0009411...
```

Registry Service is created today from master using the deploy.cmd from azure-iiot-components.
The Twin module is self built: `docker build -f docker/linux/amd64/Dockerfile -t tbn.azurecr.io/twin6 azure-iiot-opc-twin-module-master` from todays master branch and it's running on an IoTEdge device.

deployment.json:
```
"modules": {
	"publisher": {
		"type": "docker",
		"settings": {
			"image": "mcr.microsoft.com/iotedge/opc-publisher:latest",
			"createOptions": "{\"Hostname\":\"publisher\",\"Cmd\":[\"publisher\",\"--pf=./pn.json\",\"--di=60\",\"--to\",\"--aa\",\"--si=0\",\"--ms=0\"],\"ExposedPorts\":{\"62222/tcp\":{}},\"HostConfig\":{\"PortBindings\":{\"62222/tcp\":[{\"HostPort\":\"62222\"}]}}}"
		},
		"version": "1.0",
		"status": "running",
		"restartPolicy": "always"
	},
	"twin": {
		"type": "docker",
		"settings": {
			"image": "tbn.azurecr.io/twin6",
			"createOptions": "{\"NetworkingConfig\":{\"EndpointsConfig\":{\"host\":{}}},\"HostConfig\":{\"NetworkMode\":\"host\"}}"
		},
		"version": "1.0",
		"status": "running",
		"restartPolicy": "always"
	}
}
```

marcschier, 2019-07-26:
Update: We are still working on a fix.

tbn-mm, 2019-07-26:
I'm using the dev-branch (2 weeks ago) and there it's fixed.

cristipogacean, 2019-07-29:
fixed in master

marcschier, 2019-09-07:
In 2.5.1

closed 2019-09-07 by marcschier