Azure/Industrial-IoT #19: Error during publisher discovery and module execution.

CHBrianLiu, 2019-07-24:
Hi, 

I'm running IoT Edge on Windows 10 IoT Enterprise (Windows container) with OPC Twin (ver. 2.0.1) and OPC Publisher modules deployed. And I found two exception regarding to OPC Twin. Here's the detailed information of the exception. It seems like OPC twin module cannot connect to OPC Publisher: 

`[02:57:21 ERR] Error during publisher discovery.` & `[03:02:18 ERR] Error during module execution - restarting`.

May I have your suggestion to solve this problem?

Attach (1) the details of the exception and (2) module configuration for your reference.

[**Exception 1**]
```
[02:57:16 DBG] Starting Module Host...
[02:57:16 INF] Running in iotedge production context.
[02:57:21 DBG] Trying to connect to publisher...
[02:57:21 INF] Finding publisher to use as publish service...
[02:57:21 DBG] Get all modules in current edge context.
[02:57:21 DBG] Calling GET on unix:///C:/ProgramData/iotedge/workload/sock/modules?api-version=2019-01-30 uri...
[02:57:21 ERR] Error during publisher discovery.
System.Net.Internals.SocketExceptionFactory+ExtendedSocketException (10061): No connection could be made because the target machine actively refused it C
   at Microsoft.Azure.IIoT.Utils.Retry.DelayOrThrow(ILogger logger, Func`2 cont, Func`3 policy, Int32 maxRetry, Int32 k, Exception ex, CancellationToken ct)
   at Microsoft.Azure.IIoT.Utils.Retry.Do[T](ILogger logger, CancellationToken ct, Func`1 work, Func`2 cont, Func`3 policy, Int32 maxRetry)
   at Microsoft.Azure.IIoT.Module.Framework.Hosting.EdgeletClient.GetModulesAsync(String deviceId)
   at Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Servers.PublisherDiscovery.ConnectAsync()
   at Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Clients.PublisherServices.ReconnectAsync()
```

[**Exception 2**]
```
[03:01:07 INF] No publisher found - retrying in 275452 ms...
[03:02:18 ERR] Error during module execution - restarting!
System.AggregateException: One or more errors occurred. (Transient network error occurred, please retry.) ---> Microsoft.Azure.Devices.Client.Exceptions.IotHubCommunicationException: Transient network error occurred, please retry. ---> System.Net.Sockets.SocketException: A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond
   at Microsoft.Azure.Devices.Client.IotHubConnection.CreateSessionAsync(TimeSpan timeout, CancellationToken token)
   at Microsoft.Azure.Devices.Client.IotHubSingleTokenConnection.CreateSessionAsync(TimeSpan timeout, CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.FaultTolerantAmqpObject`1.OnCreateAsync(TimeSpan timeout, CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.Singleton`1.CreateValue(TaskCompletionSource`1 tcs, TimeSpan timeout, CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.Singleton`1.GetOrCreateAsync(TimeSpan timeout, CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.IotHubConnection.GetSessionAsync(TimeoutHelper timeoutHelper, CancellationToken token)
   at Microsoft.Azure.Devices.Client.IotHubConnection.CreateSendingLinkAsync(String path, IotHubConnectionString connectionString, String corrId, SendingLinkType linkType, TimeSpan timeout, ProductInfo productInfo, CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.Transport.AmqpTransportHandler.CreateEventSendingLinkAsync(TimeSpan timeout, CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.FaultTolerantAmqpObject`1.OnCreateAsync(TimeSpan timeout, CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.Singleton`1.CreateValue(TaskCompletionSource`1 tcs, TimeSpan timeout, CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.Singleton`1.GetOrCreateAsync(TimeSpan timeout, CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.Transport.AmqpTransportHandler.OpenAsync(CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.Transport.ProtocolRoutingDelegatingHandler.OpenAsync(CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.Transport.ErrorDelegatingHandler.<>c__DisplayClass22_0.<<ExecuteWithErrorHandlingAsync>b__0>d.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at Microsoft.Azure.Devices.Client.Transport.ErrorDelegatingHandler.ExecuteWithErrorHandlingAsync[T](Func`1 asyncOperation)
   --- End of inner exception stack trace ---
   at Microsoft.Azure.Devices.Client.Transport.ErrorDelegatingHandler.ExecuteWithErrorHandlingAsync[T](Func`1 asyncOperation)
   at Microsoft.Azure.Devices.Client.Transport.RetryDelegatingHandler.<>c__DisplayClass35_0.<<OpenAsyncInternal>b__0>d.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at Microsoft.Azure.Devices.Client.Transport.RetryDelegatingHandler.EnsureOpenedAsync(CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.InternalClient.OpenAsync()
   at Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory.ModuleClientAdapter.CreateAsync(String product, IotHubConnectionStringBuilder cs, String deviceId, String moduleId, ITransportSettings transportSetting, TimeSpan timeout, IRetryPolicy retry, Action onConnectionLost, ILogger logger)
   at Microsoft.Azure.IIoT.Utils.Try.Options[T](Func`1[] options)
   --- End of inner exception stack trace ---
   at Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost.StartAsync(String type, String siteId, String serviceInfo, IProcessControl reset)
   at Microsoft.Azure.IIoT.Modules.OpcUa.Twin.ModuleProcess.RunAsync() in C:\src\src\ModuleProcess.cs:line 93
---> (Inner Exception #0) Microsoft.Azure.Devices.Client.Exceptions.IotHubCommunicationException: Transient network error occurred, please retry. ---> System.Net.Sockets.SocketException: A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond
   at Microsoft.Azure.Devices.Client.IotHubConnection.CreateSessionAsync(TimeSpan timeout, CancellationToken token)
   at Microsoft.Azure.Devices.Client.IotHubSingleTokenConnection.CreateSessionAsync(TimeSpan timeout, CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.FaultTolerantAmqpObject`1.OnCreateAsync(TimeSpan timeout, CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.Singleton`1.CreateValue(TaskCompletionSource`1 tcs, TimeSpan timeout, CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.Singleton`1.GetOrCreateAsync(TimeSpan timeout, CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.IotHubConnection.GetSessionAsync(TimeoutHelper timeoutHelper, CancellationToken token)
   at Microsoft.Azure.Devices.Client.IotHubConnection.CreateSendingLinkAsync(String path, IotHubConnectionString connectionString, String corrId, SendingLinkType linkType, TimeSpan timeout, ProductInfo productInfo, CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.Transport.AmqpTransportHandler.CreateEventSendingLinkAsync(TimeSpan timeout, CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.FaultTolerantAmqpObject`1.OnCreateAsync(TimeSpan timeout, CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.Singleton`1.CreateValue(TaskCompletionSource`1 tcs, TimeSpan timeout, CancellationToken cancellationToken)
[... 49 lines cut ...]
         "--aa",
         "--si=0",
         "--ms=0"
       ],
       "ExposedPorts": {
         "62222/tcp": {}
       },
       "HostConfig": {
         "ExtraHosts": [
           "localhost:127.0.0.1"
         ],
         "Binds": [
           "c:/iiotedge:c:/appdata"
         ],
         "PortBindings": {
           "62222/tcp": [
             {
               "HostPort": "62222"
             }
           ]
         }
       }
     }

     ```

cristipogacean, 2019-07-29:
fixed in master

marcschier, 2019-09-07:
In 2.5.1

closed 2019-09-07 by marcschier