Azure/Industrial-IoT #20: Problem in start the opc twin deployment

jznling, 2019-07-24:
I followed the procedure of deploying OPC twin to IoT Edge. The docs is in 
https://docs.microsoft.com/en-us/azure/iot-accelerators/howto-opc-twin-deploy-modules

When I proceeded to start the deployment with iotedgehubdev in "run and debug locally" part, The powershell alerted with " ERROR: local variable 'module_content' referenced before assignment ". 

It seems like a coding problem. Someone made a global-local variable problem or an uninitialized variable access. 

My previous procedures are all good. There shouldn't be any problem with the other steps. 

I am not sure if I should ask you or the iotedgedev group. Please reply to me if you have any clue to fix it.

marcschier, 2019-07-31:
Duplicate of https://github.com/Azure/iotedgehubdev/issues/195

marcschier, 2019-07-31:
Duplicate of https://github.com/Azure/iotedgehubdev/issues/183

closed 2019-07-31 by marcschier