Azure/Industrial-IoT #21: Communication error when browsing endpoint (Running in IoT Edge)

ryan-ms, 2019-07-24:
Hi, I followed the instruction [here]( https://docs.microsoft.com/en-us/azure/iot-accelerators/howto-opc-twin-deploy-modules) to deploy the opc-twin module in IoT edge.

And I can successfully discover my OPC UA server, and find endpoint, but when I started to browse node, the OPC-twin module shows error as below.

[09:32:16 DBG] uat4600fc959c1d15666910d6166ef5c0f76203de0d twin starting...
[09:32:16 DBG] Starting Module Host...
[09:32:16 INF] Running in iotedge production context.
[09:32:17 INF] Module moxagateway_opctwin connection status changed to Connected due to Connection_Ok.
[09:32:17 INF] Module moxagateway_opctwin connection status changed to Disconnected_Retrying due to Communication_Error.
[09:32:17 INF] Module moxagateway_opctwin connection status changed to Disconnected_Retrying due to Communication_Error.
[09:32:17 INF] Module moxagateway_opctwin connection status changed to Connected due to Connection_Ok.
[09:32:17 ERR] Error #1 in twin host - restarting...
System.TimeoutException: The operation timed out. ---> System.Threading.Tasks.TaskCanceledException: A task was canceled.
   at Microsoft.Azure.Devices.Client.Transport.AmqpTransportHandler.RoundTripTwinMessage(AmqpMessage amqpMessage, CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.Transport.AmqpTransportHandler.SendTwinGetAsync(CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.Transport.ErrorDelegatingHandler.ExecuteWithErrorHandlingAsync[T](Func`1 asyncOperation)
   at Microsoft.Azure.Devices.Client.Transport.RetryDelegatingHandler.<>c__DisplayClass27_0.<<SendTwinGetAsync>b__0>d.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at Microsoft.Azure.Devices.Client.Transport.RetryDelegatingHandler.SendTwinGetAsync(CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.InternalClient.GetTwinAsync()
   --- End of inner exception stack trace ---
   at Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost.StartAsync(String type, String siteId, String serviceInfo, IProcessControl reset)
   at Microsoft.Azure.IIoT.OpcUa.Edge.Supervisor.SupervisorServices.TwinHost.RunAsync()

[image]

But It will not happen while I run the twin-module via docker-compose.
Do you have any idea to slove the problem? thank you.

ryan-ms, 2019-07-25:
Hi @karok2m 

Thanks for sharing. Looking forward the updated version and document.

ryan-ms, 2019-07-30:
Hi, I've noticed the master is fixed.
Haven't try it, but using the `mcr.microsoft.com/iotedge/opc-twin:latestare` is still not working.
Is the mcr image already updated?

marcschier, 2019-08-09:
This will be in release 2.5.0, tentative mid-august

marcschier, 2019-09-07:
In 2.5.1.

closed 2019-09-07 by marcschier