Azure/Industrial-IoT #26: OPC Tag not showing when browsing node in some PLC with OPC UA Server

ryan-ms, 2019-07-25:
I built the OPC Twin module from `develop` branch, and use Connected factory to test the discover, read, write.

And it work fine on my Siemens S7-1500 PLC.
But when I changed to another Siemens S7-1500 PLC (but with different CPU module).

It cannot shows all tag when I expanded the node.
Like picture below, it only shows "icon"

[image]

But when I use other OPC client ( OPC Foundation Sample Client )to connect to PLC and browse the OPC node. it can successfully show all tags.

[image]

But before showing all the tag, it prompt a window with "more reference exists" message when I clicked "memory" which not happened whenI connected to other OPC UA server.

[image]

There are sightly different on CPU module in PLC, like table below.

PLC | CPU | Result
-- | -- | --
Siemens-S7 | CPU   1511-1 PN6ES7511-1AK01-0AB0 | Success       Can Browse        all OPC tag (V)  
Siemens-S7 | CPU   1511-1 PN6ES7511-1AK02-0AB0 | Failed   -  Cannot Browse   all OPC tag,    Only show   “icon”

Not sure is the CPU matter or the incorrect setting in PLC? 
Any idea?

marcschier, 2019-07-26:
Hi @ryan-ms, we have seen several similar issues in our lab testing, and are looking at root causing it.

cristipogacean, 2019-07-29:
Hi @ryan-ms, I'm not sure if this is an incorrect setting in the PLC, nevertheless I think the opctwin should be able to display the address space consistently like other clients. So we would like to know a bit more about your setup:
 1) what version of the opctwin are you using?
 2) what are the node IDs of the items that are not displayed (just 2-3 of them)?
 3) can you share with us the namespace array of the server?

ryan-ms, 2019-07-30:
Hi @cristipogacean ,

Thanks for your reply, let me provide the detail.

1. I used the opc twin image which I download and build from develop branch on 7/23.
   Here is my image: `ryan77119/azure-iiot-opc-twin-module:latest`
   It's a linux container running in ubuntu.

2. NodeId:
    Sensor_100_ON: "ns=3;s="Sensor_I00_ON"
    [image]

     Sensor_IW2_PV: "ns=3;s="Sensor_IW2_PV"
    [image]

     Sensor_IW2_SCALE: ""Sensor_IW2_SCALE"
    [image]

3. The namespace array are:
    [0]: http://opcfoundation.org/UA/
    [1]: urn:SIMATIC.S7-1500.OPC-UAServer:PLC_1
    [2]: http://opcfoundation.org/UA/DI/
    [3]: http://www.siemens.com/simatic-s7-opcua

[image]

Thank you very much. Looking forward your reply.

marcschier, 2019-08-09:
This looks like an issue in connected factory and a naming/functionality issue in the API.   

Any continuation point reported by the server as result of browse is passed back to the client and client should call BrowseNext with same continuation point.   However, connected factory browse controller uses only `NodeBrowseAsync`, which is actually `NodeBrowseFirstAsync`.   

I propose to rename the original NodeBrowseAsync on the API interface to NodeBrowseFirstAsync to be in line with the business logic service API, and then add an extension method called NodeBrowseAsync which browses the entire set of references - same as the service extension.

marcschier, 2019-09-07:
Released in 2.5.1.  It is not yet in CF but you can use the CLI to validate.

ryan-ms, 2019-09-10:
Thanks @marcschier , I will try it.
But how can I know when is the CF updated?  Can you leave message here while the CF updated?
Thanks a lot.

ryan-ms, 2019-09-12:
Hi @marcschier ,

I still met the same issue while using latest opc twin service and module.

I've download the latest code yesterday (9/11) and follow the latest document ,[Deploying Azure Industrial IoT Microservices and dependencies](https://github.com/Azure/Industrial-IoT/blob/master/docs/howto-deploy-microservices.md) instructions to deploy service to Azure

and follow [Deploy OPC Twin and OPC Publisher Edge Modules](https://github.com/Azure/Industrial-IoT/blob/master/docs/howto-deploy-modules.md) instruction to deploy the opc twin and publisher module to Linux gateway.

Then I use the opc twin react web app to discover, browse the OPC UA Server.
But there is the same issue...

Like the following picture, I can see servals OPC tag in Tags node
[image]

But cannot see them in the web app.
[image]

marcschier, 2019-09-12:
Please use the command line interface to browse and see if the results match the other client, it should.  We did not update the sample app not connected factory, both do not use continuations properly.  The CLI however does.

Holen Sie sich Outlook für Android<https://aka.ms/ghei36>

[Client]<https://nam06.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F29178018%2F64751354-6f1f5300-d54e-11e9-909a-f0687abeeff6.png&data=02%7C01%7CMarc.Schier%40microsoft.com%7C0c1af954a51d4b9768a708d73730cd86%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C637038555053453201&sdata=0lVZjD%2BdnZ20NxjGpNDaRX%2BAESi%2FfqoND8bdAaF9g8c%3D&reserved=0>

[web]<https://nam06.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F29178018%2F64751522-ee148b80-d54e-11e9-822f-61de3a6a8a60.png&data=02%7C01%7CMarc.Schier%40microsoft.com%7C0c1af954a51d4b9768a708d73730cd86%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C637038555053453201&sdata=EkMeulJ1kZ9%2F97Vy%2BQR%2BeW2h64XmNCXSqCyYOixad14%3D&reserved=0>

ryan-ms, 2019-09-12:
> Hi @marcschier ,
> 
> I still met the same issue while using latest opc twin service and module.
> 
> I've download the latest code yesterday (9/11) and follow the latest document ,[Deploying Azure Industrial IoT Microservices and dependencies](https://github.com/Azure/Industrial-IoT/blob/master/docs/howto-deploy-microservices.md) instructions to deploy service to Azure
> 
> and follow [Deploy OPC Twin and OPC Publisher Edge Modules](https://github.com/Azure/Industrial-IoT/blob/master/docs/howto-deploy-modules.md) instruction to deploy the opc twin(mcr.microsoft.com/iotedge/opc-publisher:latest) and publisher module (mcr.microsoft.com/iotedge/opc-publisher:latest) to Linux gateway.
> 
> Then I use the opc twin react web app to discover, browse the OPC UA Server.
> But there is the same issue...
> 
> Like the following picture, I can see servals OPC tag in Tags node
> [image]
> 
> But cannot see them in the web app.
> [image]

I think that's a different issue from original issue.
I just test the Siemens PLC which I tested before, and notice the issue is resolved and works very well, thank you.

And for the new issue in other PLC seems like caused by  bad nodeId error without error handler. 
Do you have any idea?

ryan-ms, 2019-09-12:
> Please use the command line interface to browse and see if the results match the other client, it should. We did not update the sample app not connected factory, both do not use continuations properly. The CLI however does. Holen Sie sich Outlook für Android<https://aka.ms/ghei36>
> […](#)

Thanks, Let  me try the CLI first, thank you.

closed 2019-09-26 by marcschier