Azure/Industrial-IoT #27: Deployment script for microservices fails

michaellindberg, 2019-07-25:
Hi,

When I try to follow the guideline for deploying the microservices I run into several issues and I had to do some changes to the https://github.com/Azure/Industrial-IoT/blob/master/services/deploy/vm/template.json file to get it to a stage where it starts deploying components in Azure. But I can't get it to completely succeed, I hope you can help.

1. There is an issue with a Keyvault resource pointing to a variable named keyVaultVersion which does not exist under variables (I changed this to keyVaultApiVersion) which will remove this error.
2. It seems that the script tries to create two key vaults with the same name. Removing the key vault below solves the duplicate key vault issue and the resources will start to deploy to Azure.
`{
            "type": "Microsoft.KeyVault/vaults",
            "name": "[parameters('keyVaultName')]",
            "apiVersion": "[variables('keyVaultVersion')]",
            "location": "[resourceGroup().location]",
            "tags": {
                "displayName": "[parameters('keyVaultDisplayName')]"
            },
            "properties": {
                "enabledForDeployment": true,
                "enabledForTemplateDeployment": false,
                "enabledForVolumeEncryption": false,
                "sku": {
                    "name": "[parameters('keyVaultSkuName')]",
                    "family": "A"
                },
                "tenantId": "[parameters('aadTenantId')]",
                "accessPolicies": []
            }
        },`
3. A BadRequest error is received when trying to create the key vault secret `/PCS_IOTHUBREACT_AZUREBLOB_KEY`

I tried to follow this [guide](https://github.com/Azure/Industrial-IoT/blob/master/docs/howto-deploy-microservices.md#prerequisites) both with the old repo https://github.com/Azure/azure-iiot-components, the new repo https://github.com/Azure/Industrial-IoT master branch and the "new_repo_structure" branch.

Thanks :)

marcschier, 2019-07-26:
Hi @michaellindberg, thanks for filing the issue :-)

Apologies for you hitting multiple problems here, among them:

* We renamed the repo and removed the submodules to create a mono-repo, and thus the arm template pulling the VM contents failed in multiple places.
* The key vault api version variable in the ARM template has probably fallen victim in a manual merge.
* The secret storage from ARM is broken.
* We moved from classic to azure-pipelines.yml, which is not fully up and operational yet and thus haven't caught any of this.

I am working on a fix now and should have deployment working again by end of day GMT.  Manually validated.

marcschier, 2019-07-26:
Fixed with #28 - in lieu of deployment being part of CI at this point (another work item) I manually tested cloud and local deployment.   Please try out and re-open if you see issues.

michaellindberg, 2019-07-29:
I can confirm that the deploy script works now. I get a 404 when accessing the website (https://_solutionname_.azurewebsites.net/), but I will try and have a look at it before opening a new issue.

closed 2019-07-29 by michaellindberg