Azure/Industrial-IoT #32: Linux Image

ryan-ms, 2019-07-31:
Hi,

I used the latest source code on master branch to build opc twin module today.

I used  the dockerFile in `Industrial-IoT/modules/opc-twin/docker/Linux/amd64/`to build the linux image and running in IoT edge on Ubuntu OS.

But there are some error while scanning the network.
It show the error log as below in OPC twin log.

`[08:17:12 ERR] (Error) Channel 76685: ProcessErrorMessage(BadNotFound 'Error received from remote host: ')`

But it still shows the result uploaded.
[image]

But after that, when I click the endpoint button to see the endpoint. there are no any endpoint in the endpoint table, like picture below.
[image]

I've tried the image which build at last week in develop branch, but the same error not happened, it works fine with that version.

Thank you.

cristipogacean, 2019-07-31:
Thanks for reporting this. We are working on fixing it.

monikavar, 2020-02-24:
Hi @ryan-ms , I have tested with the latest code on master and scenario for discovering endpoints running IoT Edge on Ubuntu OS seems to work (BTW we don't use dockerFile to create linux image anymore). Could you please test it with the latest code on master and re-open this issue in case of any problems?

closed 2020-02-24 by monikavar

ryan-ms, 2020-02-25:
Hi @monikavar,
I've tested the latest module. And it ran perfectly. Thank you!