Azure/Industrial-IoT #36: Cannot Capture the ServerTimestamp by using the telemetry configuration

aakash1024, 2019-08-03:
Hi ,

I want to capture the ServerTimestamp of the OPCUA server as part of my configuration . Have used the telemetry json as below but i am unable to get the ServerTimestamp in my iot Stream. can some one confirm if this is the desired configuration or something has to be changed.
`{ "Defaults": { "Flat": true, "Value": { "Flat": true, "Value": { "Publish": true, "Name": "Value" }, "SourceTimestamp": { "Publish": true, "Name": "SourceTimestamp" }, "ServerTimestamp": { "Publish": true, "Name": "ServerTimestamp" }, "StatusCode": { "Publish": true, "Name": "StatusCode" // 'Pattern' is ignored for the 'StatusCode' value }, // 'Status' is the symbolic name of 'StatusCode' "Status": { "Publish": false, "Name": "Status" } } } }`

WaywardHayward, 2019-09-11:
We are having this issue as well - how can we get the "serverTimestamp" to be sent as well as the "sourceTimestamp" with the value on a monitored OPC item.

I guess this would involve a change to IMessageData and to the method "MonitoredItemNotificationEventHandler" 
something like this [snippet](https://github.com/Azure/Industrial-IoT/blob/be4bf5c6ae6f5be23a9e1edbbb2cf06b2a9ff7e0/modules/opc-publisher/opcpublisher/OpcMonitoredItem.cs#L355)

Is there any chance this is in the pipeline?

Thanks

cristipogacean, 2020-03-29:
This is fixed with the publisher refactoring as part of the OPC UA Pub/Sub compliance, please check: https://github.com/Azure/Industrial-IoT/blob/master/docs/dev-guides/telemetry-messages-format.md

Please reopen if you need more support.

closed 2020-03-29 by cristipogacean