Azure/Industrial-IoT #38: OPC UA PubSub 1.04 Compliance

mpostol, 2019-08-03:
The Readme.md file states:

> This reference implementation demonstrates how Azure IoT Edge can be used to connect to existing OPC UA servers and publishes JSON encoded telemetry data from these servers in OPC UA "Pub/Sub" format (using a JSON payload) to Azure IoT Hub. 

It requires that the message created by:

`IotHubMessaging.CreateJsonMessageAsync(OpcPublisher.OpcMonitoredItem.MessageData)`

has to be compliant with the specification:

- [OPC Unified Architecture Specification Part 14: PubSub Release 1.04 February 06, 2018][OPC.UA.PubSub]

My concern is what is the reason to create the message following this specification. As far as I understand this encoding. i.e. the JSON based message mapping allows OPC UA Applications to interoperate with web and enterprise software that use this format. Is there any plan to:
- make the Azure IoT Edge OPC UA PubSub JSON encoding aware
- compliant with OPC UA PubSub Application and offer the Subscriber role.

[OPC.UA.PubSub]: https://opcfoundation.org/developer-tools/specifications-unified-architecture/part-14-pubsub/

mpostol, 2020-01-12:
To promote reusability this issue must be reviewed against the mpostol/OPC-UA-OOI#425 and new project [Object-Oriented Internet - reactive visualization of asynchronous data using AZURE services](https://github.com/mpostol/OPC-UA-OOI/projects/15).

marcschier, 2020-02-19:
We are working on Pub/Sub format and will release a configuration API before June this year.

mpostol, 2020-02-19:
@marcschier Consider using my document that [contains Augmented BNF definition](https://github.com/mpostol/OPC-UA-OOI/blob/master/Networking/SemanticData/MessageHandling/NetworkMessage.abnf) of the NetworkMessage. NetworkMessage is originally defined in OPC UA Part 14  Release 1.04 February 06, 2018.   Augmented BNF is defined in the document Augmented BNF for Syntax Specifications: ABNF RFC 5234. 

OPCF defined also a document  `OPC 10001-6 - Amendment 6 UADP Header Layouts`. 
There are also reported issues in the Mantis. 
As far as I know, OPCF does not have any compliance procedures.
My concern is if all of that could be used as a consistent foundation for implementation promoting interoperability.

mpostol, 2020-04-03:
Hi @cristipogacean, 

FYI I try to figure out where/how PubSub is implemented.  Check out the results of my investigation [here](https://github.com/mpostol/OPC-UA-OOI/issues/208#issuecomment-608511722).  
Let me know if you know other implementations. 

To get more about my approach check out my article [Object-Oriented Internet Reactive Interoperability](https://www.researchgate.net/publication/336020749_Object-Oriented_Internet_Reactive_Interoperability).

Let me know how I can help.

rowenwu, 2020-06-29:
Hi @mpostol, 
Could you describe your use case so I can better understand what you're currently working on in OOI and OPC UA pubsub. Maybe it would make more sense to get on a call and discuss in depth. Feel free to reach out over email at rowwu@microsoft.com

mpostol, 2020-09-24:
Hi @barnstee and @mregen, I understand that removing this issue from the 2.8 milestone means that compatibility with OPC UA PubSub 1.04 is not your priority for now. Let me inform you that [PubSub to Azure Gateway](https://github.com/mpostol/OPC-UA-OOI/tree/master/Networking/DataRepository/AzureGateway#azure-gateway-datarepository-implementation) has been just published on GitHub as a part of the more general concept [Object-Oriented Internet](https://github.com/mpostol/OPC-UA-OOI) reactive networking. Hopefully, at some point, we will have the opportunity to test against the interoperability of our implementations. Of course, it makes sense only if our solutions will be compliant with a selected specification release (not sure if the current release is a perfect match).

Because the compliance with the OPC UA 1.04 is also no longer important for me you may close this issue. We may open a new one, in any case, there will be any need to harmonize our work.

I will appreciate keeping me informed about your plans in this respect.

barnstee, 2020-09-24:
You are making a lot of assumptions here! We just removed the milestone since we're not sure yet if we call the next version 2.8 or 3.0. This work is actually scheduled for our current sprint! :)

mpostol, 2020-09-24:
@barnstee, it is great. I only assumed that the "sprint a set period of time during which specific tasks must be completed" but not tightly coupled version number". Thanks, I will watch it. Let me know if there is something I can do to help.