Azure/Industrial-IoT #39: JSON message format to IotHub

davebanthorpe, 2019-08-03:
Hi - I have the module up and running pulling data from a BACnet device (via Kepware). The format of the output JSON is as follows:

{
  "NodeId": "nsu=KEPServerEX;s=Channel1.Device2.AnalogInput_1.Water Temperature",
  "ApplicationUri": "urn:DAVEDEV:Kepware.KEPServerEX.V6:UA%20Server",
  "DisplayName": "Water Temperature",
  "Value": 37.1,
  "SourceTimestamp": "2018-04-30T14:09:59.4983294Z"
},{
  "NodeId": "nsu=KEPServerEX;s=Channel1.Device2.AnalogInput_0.Indoor Temperature",
  "ApplicationUri": "urn:DAVEDEV:Kepware.KEPServerEX.V6:UA%20Server",
  "DisplayName": "Indoor Temperature",
  "Value": 20.9,
  "SourceTimestamp": "2018-04-30T14:10:00.4993239Z"
},
........

How do I configure it to send it as the following format (i.e. merging the DisplayName and Value attributes into a single parameter) so that I can via the streaming data in PowerBI?

{
  "NodeId": "nsu=KEPServerEX;s=Channel1.Device2.AnalogInput_1.Water Temperature",
  "ApplicationUri": "urn:DAVEDEV:Kepware.KEPServerEX.V6:UA%20Server",
  "Water Temperature": "37.1",
  "SourceTimestamp": "2018-04-30T14:09:59.4983294Z"
},{
  "NodeId": "nsu=KEPServerEX;s=Channel1.Device2.AnalogInput_0.Indoor Temperature",
  "ApplicationUri": "urn:DAVEDEV:Kepware.KEPServerEX.V6:UA%20Server",
  "Indoor Temperature": "20.9",
  "SourceTimestamp": "2018-04-30T14:10:00.4993239Z"
},
.......

marcschier, 2020-02-19:
Hi, we are moving to PubSub (Json/Binary) message format (Network message) - which is defined in OPC UA Part 14.   As such we are not going to document the old legacy format, however, the decoding code is part of the platform - you can find it in MonitoredItemMessageApiModel.cs.

closed 2020-02-19 by marcschier

jlorich, 2020-06-22:
@hansgschossmann I see --ic is no longer available in the publisher 2.6 and above.  Is there a recommended alternative for getting data in a format that is compatible with DTDL (e.g. IoT Central)? This is going to be particularly problematic with the upcoming release of Azure Digital Twins.

dvanlaerhoven, 2020-09-19:
@hansgschossmann @marcschier Is there any news on getting the data format compatible with DTDL and IoT central?

aronsemle, 2020-10-02:
I heard rumor something was coming with DTDL and the edge stuff, but I'm an old Kepware guy and very familiar with this problem. We're developing something to make this easier (basically OPC UA to nicely modeled JSON objects in Azure), among ither things. It's called HighByte (https://www.highbyte.com/). It's early days, so ping me if you're interested in trying it out.