Azure/Industrial-IoT #40: OPC Publisher won't send messages again after a lost connection

tbuquet, 2019-08-03:
Hello,

I'm currently having an issue with OPC Publisher. If for any reason the connection is lost between OPC Publisher and a OPC Server, it will attempt to reconnect (and seems to do so successfully), but then OPC Publisher won't receive node updates anymore and won't send anything back to IoT Hub.

Here's the diagnostic:

`5/5/2018

 00:15:32.470 ==========================================================================
5/5/2018 00:16:32.479 ==========================================================================
5/5/2018 00:16:32.482 OpcPublisher status @ 5/5/2018 5:16:32 AM (started @ 5/4/2018 8:02:08 PM)
5/5/2018 00:16:32.484 ---------------------------------
5/5/2018 00:16:32.486 OPC sessions: 2
5/5/2018 00:16:32.487 connected OPC sessions: 2
5/5/2018 00:16:32.489 connected OPC subscriptions: 2
5/5/2018 00:16:32.490 OPC monitored items: 4466
5/5/2018 00:16:32.492 ---------------------------------
5/5/2018 00:16:32.494 monitored items queue bounded capacity: 8192
5/5/2018 00:16:32.495 monitored items queue current items: 0
5/5/2018 00:16:32.497 monitored item notifications enqueued: 83489
5/5/2018 00:16:32.499 monitored item notifications enqueue failure: 0
5/5/2018 00:16:32.500 monitored item notifications dequeued: 83489
5/5/2018 00:16:32.502 ---------------------------------
5/5/2018 00:16:32.503 messages sent to IoTHub: 4978
5/5/2018 00:16:32.504 last successful msg sent @: 5/5/2018 4:22:28 AM
5/5/2018 00:16:32.506 bytes sent to IoTHub: 19684740
5/5/2018 00:16:32.507 avg msg size: 3954
5/5/2018 00:16:32.509 msg send failures: 0
5/5/2018 00:16:32.511 messages too large to sent to IoTHub: 0
5/5/2018 00:16:32.512 times we missed send interval: 167
5/5/2018 00:16:32.514 ---------------------------------
5/5/2018 00:16:32.516 current working set in MB: 89
5/5/2018 00:16:32.518 --si setting: 5
5/5/2018 00:16:32.519 --ms setting: 262144
5/5/2018 00:16:32.521 --ih setting: Amqp_Tcp_Only
5/5/2018 00:16:32.523 ==========================================================================
5/5/2018 00:16:54.241 SECURE CHANNEL RENEWED [.NetStandard ServerChannel UA-TCP 1.03.351UA-TCP 1.03.351] [ID=5]
5/5/2018 00:17:25.580 Session endpoint: opc.tcp://x.x.x.x:4096/iaopcua/Basic128Rsa15 has Status: BadNoCommunication 'Server not responding to keep alive requests.'
5/5/2018 00:17:25.580 Session endpoint: opc.tcp://x.x.x.x:49320/ has Status: BadNoCommunication 'Server not responding to keep alive requests.'
5/5/2018 00:17:25.604 Session endpoint: opc.tcp://x.x.x.x:49320/ has Status: BadNoCommunication 'Server not responding to keep alive requests.'
5/5/2018 00:17:25.618 Session endpoint: opc.tcp://x.x.x.x:49320/ has Status: BadNoCommunication 'Server not responding to keep alive requests.'
5/5/2018 00:17:25.622 Session endpoint: opc.tcp://x.x.x.x:4096/iaopcua/Basic128Rsa15 has Status: BadNoCommunication 'Server not responding to keep alive requests.'
5/5/2018 00:17:25.703 ***EXCEPTION*** BadNoSubscription
5/5/2018 00:17:25.998 Outstanding requests: 0, Defunct requests: 0
5/5/2018 00:17:26.041 Outstanding requests: 0, Defunct requests: 0
5/5/2018 00:17:26.055 Outstanding requests: 0, Defunct requests: 0
5/5/2018 00:17:26.080 Outstanding requests: 0, Defunct requests: 0
5/5/2018 00:17:26.103 Outstanding requests: 0, Defunct requests: 0
5/5/2018 00:17:26.124 Good publish requests: 0, KeepAlive interval: 2000
5/5/2018 00:17:26.141 Good publish requests: 0, KeepAlive interval: 2000
5/5/2018 00:17:26.157 Good publish requests: 0, KeepAlive interval: 2000
5/5/2018 00:17:26.167 Good publish requests: 0, KeepAlive interval: 2000
5/5/2018 00:17:26.213 SessionId: ns=1;g=bc5cf924-1457-4cc2-b0b2-45f314acfde4
5/5/2018 00:17:26.181 Good publish requests: 0, KeepAlive interval: 2000
5/5/2018 00:17:26.224 SessionId: ns=1;i=3
5/5/2018 00:17:26.228 Missed KeepAlives: 1
5/5/2018 00:17:26.204 SessionId: ns=1;g=bc5cf924-1457-4cc2-b0b2-45f314acfde4
5/5/2018 00:17:26.190 SessionId: ns=1;i=3
5/5/2018 00:17:26.203 SessionId: ns=1;g=bc5cf924-1457-4cc2-b0b2-45f314acfde4
5/5/2018 00:17:26.220 Missed KeepAlives: 1
5/5/2018 00:17:26.241 Session endpoint: opc.tcp://10.4.13.63:4096/iaopcua/Basic128Rsa15 got a keep alive after 1 was missed.
5/5/2018 00:17:26.248 Missed KeepAlives: 2
5/5/2018 00:17:26.257 Missed KeepAlives: 2
5/5/2018 00:17:26.277 Session endpoint: opc.tcp://x.x.x.x:49320/ got a keep alive after 3 were missed.
5/5/2018 00:17:26.294 Session endpoint: opc.tcp://x.x.x.x:49320/ got a keep alive after 3 were missed.
5/5/2018 00:17:26.267 Missed KeepAlives: 3
5/5/2018 00:17:26.293 ***EXCEPTION*** BadNoSubscription
5/5/2018 00:17:32.513 ==========================================================================
5/5/2018 00:17:32.517 OpcPublisher status @ 5/5/2018 5:17:32 AM (started @ 5/4/2018 8:02:08 PM)
5/5/2018 00:17:32.518 ---------------------------------
5/5/2018 00:17:32.520 OPC sessions: 2
5/5/2018 00:17:32.521 connected OPC sessions: 2
5/5/2018 00:17:32.523 connected OPC subscriptions: 2
5/5/2018 00:17:32.525 OPC monitored items: 4466
5/5/2018 00:17:32.526 ---------------------------------
5/5/2018 00:17:32.528 monitored items queue bounded capacity: 8192
5/5/2018 00:17:32.530 monitored items queue current items: 0
5/5/2018 00:17:32.531 monitored item notifications enqueued: 83489
5/5/2018 00:17:32.533 monitored item notifications enqueue failure: 0
5/5/2018 00:17:32.534 monitored item notifications dequeued: 83489
5/5/2018 00:17:32.536 ---------------------------------
5/5/2018 00:17:32.537 messages sent to IoTHub: 4978
5/5/2018 00:17:32.540 last successful msg sent @: 5/5/2018 4:22:28 AM
5/5/2018 00:17:32.542 bytes sent to IoTHub: 19684740
5/5/2018 00:17:32.544 avg msg size: 3954
5/5/2018 00:17:32.545 msg send failures: 0
5/5/2018 00:17:32.547 messages too large to sent to IoTHub: 0
5/5/2018 00:17:32.549 times we missed send interval: 168
5/5/2018 00:17:32.550 ---------------------------------
5/5/2018 00:17:32.553 current working set in MB: 89
5/5/2018 00:17:32.555 --si setting: 5
5/5/2018 00:17:32.557 --ms setting: 262144
5/5/2018 00:17:32.559 --ih setting: Amqp_Tcp_Only
5/5/2018 00:17:32.560 ==========================================================================
5/5/2018 00:18:32.569 ==========================================================================
5/5/2018 00:18:32.576 OpcPublisher status @ 5/5/2018 5:18:32 AM (started @ 5/4/2018 8:02:08 PM)
5/5/2018 00:18:32.578 ---------------------------------
5/5/2018 00:18:32.580 OPC sessions: 2
5/5/2018 00:18:32.582 connected OPC sessions: 2
5/5/2018 00:18:32.584 connected OPC subscriptions: 2
5/5/2018 00:18:32.586 OPC monitored items: 4466
5/5/2018 00:18:32.588 ---------------------------------
5/5/2018 00:18:32.589 monitored items queue bounded capacity: 8192
5/5/2018 00:18:32.591 monitored items queue current items: 0
5/5/2018 00:18:32.593 monitored item notifications enqueued: 83489
5/5/2018 00:18:32.595 monitored item notifications enqueue failure: 0
5/5/2018 00:18:32.596 monitored item notifications dequeued: 83489
5/5/2018 00:18:32.598 ---------------------------------
5/5/2018 00:18:32.599 messages sent to IoTHub: 4978
5/5/2018 00:18:32.601 last successful msg sent @: 5/5/2018 4:22:28 AM
5/5/2018 00:18:32.603 bytes sent to IoTHub: 19684740
5/5/2018 00:18:32.604 avg msg size: 3954
5/5/2018 00:18:32.606 msg send failures: 0
5/5/2018 00:18:32.607 messages too large to sent to IoTHub: 0
5/5/2018 00:18:32.609 times we missed send interval: 168
5/5/2018 00:18:32.611 ---------------------------------
5/5/2018 00:18:32.613 current working set in MB: 90
5/5/2018 00:18:32.615 --si setting: 5
5/5/2018 00:18:32.617 --ms setting: 262144
5/5/2018 00:18:32.619 --ih setting: Amqp_Tcp_Only
5/5/2018 00:18:32.621 ==========================================================================`

I'm not entirely sure what's going on, it seems to get back the keepalive messages after a connection was dropped, 5 being the max amount of keepalive to send before the session is disconnected. Any idea what might be causing this issue?

Thanks!

MarekLani, 2019-12-19:
Please any update on this one?

markusstuhler, 2020-02-20:
Hi @MarekLani ,

we fixed the BadNoSubscription-Error in 2.5.3. Please try that version!
I'm closing this issue for now. Please reopen if the issue persists with 2.5.3

Thanks!
Markus

closed 2020-02-20 by markusstuhler