Azure/Industrial-IoT #41: Using username/password instead of certificate to get authenticated to an OPC UA server

kotselio, 2019-08-03:
Hello! It's often the case that an OPC UA server does not have a certificate, but only username/password type of credentials. AFAIU there is no support for this feature in the current version of the publisher. Any chance that it's gonna be added?

lucarv, 2019-09-02:
do we have any updates on this? any work around considering the OPC-UA server i am trying to connect to is password protected?

markusstuhler, 2019-12-03:
Hi,

this is implemented as described here: https://docs.microsoft.com/en-us/azure/iot-accelerators/howto-opc-publisher-configure#configure-username-and-password-for-authentication

Thanks!
Markus

closed 2019-12-03 by markusstuhler

MasterR8, 2019-12-10:
Is there no way to use this feature with the publishednodes.json? I always get the following error:
System.NullReferenceException: Could not retrieve credentials to authenticate to the server. Please check if 'OpcAuthenticationUsername' and 'OpcAuthenticationPassword' are set in configuration.

My Config:
...
"OpcAuthenticationMode": "UsernamePassword",
"OpcAuthenticationUsername": "<username>"
"OpcAuthenticationPassword": "<password>"
...