Azure/Industrial-IoT #44: What about the publication of Events? (feature request)

eoursel, 2019-08-03:
The OPC UA PubSub specification supports the publication of OPC UA Events. In the Part 14 specification, there is the PublishedEventsType. 

Do you plan to support the publication of events in the OPC Publisher? 

As a matter of fact, most industrial processes generate Alarms. This feature is really important for applications which need to analyze and store alarms with Azure services. 

Best regards,

rowenwu, 2020-06-29:
Hi @eoursel, I'm the PM on the team prioritizing our backlog, and I'd love to learn more about your request for Alarms/Events support and what you're looking to do with it. This sort of information helps us evaluate customer needs. Please reach out via email to rowwu@microsoft.com if you'd like to chat further.

rowenwu, 2020-06-29:
And @strandfelt, @dmakeranko-connyun, @boske001, @bewacontrol-renedivossen, since you've also shown interest in this feature, please do feel free to reach out as well! Send me a note at rowwu@microsoft.com

MkTru, 2020-10-30:
Any update on Alarms/Events support?

nelsonmorais, 2021-08-03:
I'm also looking for an update on this request. There's also another request #1219 for the same feature.

closed 2022-12-12 by marcschier