Azure/Industrial-IoT #49: Communication with IoT hub stops after some hours.

anhame, 2019-08-03:
Hello,
I'm using the OpcPublisher with my IoT hub as a standalone application and the Mqtt protocol.
My problem is that after a few hours of operation I'm receiving the following error:
**Connection status changed to 'Disconnected', reason 'Retry_Expired'**
After this, the communication with IoT hub is completely lost and the msg send failures counter in the OpcPublisher status output starts to increase.
Before there were also several messages (repeated almost hourly) like this:
**Connection status changed to 'Disconnected_Retrying', reason 'Communication_Error'**

What could be reason for this? There were no such problems occurring in former versions of the publisher (half a year ago).

Any help appreciated.
Thanks
Hartmut

YashC, 2019-09-15:
Any solution to this issue?

anhame, 2019-09-16:
Unfortunately not. I did a support request to MS support/IoT Hub. Found out that it only occurs when using mqtt-websocket-only and amqp-websocket-only. Works with Amqp.
MS support told me to clarify this directly with the owners of the opc publisher...
Still hoping to get some help here since supporting communication over websocket is required in many of our installations.
regards
Hartmut

marcschier, 2019-09-18:
Hi @anhame - apologies for this dragging out so long and us not getting back.  We are building a tool to track these issues to the correct owner (Us, IoT SDK or IoT Edge).  

While we do this, want to clarify the working setup where raw mqtt and amqp works, does this mean connected over raw mqtt or amqp to EdgeHub, or to IoTHub?   And if the former, is EdgeHub connected over Websocket?   Or also over raw MQTT or AMQP?

anhame, 2019-09-18:
Hi Marc,
the connection setup is just raw AMQP to IoTHub (didn't test with raw MQTT so far), it doesn't work over Websocket (neither AMQP nor MQTT).
regards

marcschier, 2019-09-18:
Further clarification - this is publisher connected directly to IoT Hub , not running inside IoT Edge, correct?

anhame, 2019-09-18:
Yes, the opc publisher running as a standalone app connected directly to the IoT hub.

anhame, 2019-09-18:
If you wish, I could add the last lines of the publisher logs with detailed Device SDK output (containing the exception that causes the communication to be stopped)?

marcschier, 2019-09-18:
That would be fantastic.   Btw., which version of publisher are you using, is it 2.5.1 or earlier?   So we know the version of IoT sdk in question also.

anhame, 2019-09-18:
I'm using 
publisher version 2.4.1,
Microsoft.Azure.Devices 1.17.3,
Microsoft..Azure.Devices.Client 1.20.1
I've attached the complete log file since it could also be helpful to see the complete history...
[publisherLogs.zip](https://github.com/Azure/Industrial-IoT/files/3625788/publisherLogs.zip)

marcschier, 2019-09-26:
Using a tool to introduce bad packets into the network I can reliable get websocket communication to crash, not just in publisher, but in general.   Following up with SDK team.  To confirm - the issue does not repro for you if not using websocket, so this could be the smoking gun.

anhame, 2019-09-27:
Yes, the publisher connectivity is fine when using Amqp...

marcschier, 2020-03-29:
Closing as we officially only support running our modules in IoT edge and thus websocket is not required.  

Please open an issue with the azure iot sdk if you want to pursue issues in websocket connectivity or consider running with mqtt or amqp against IoT edge in transparent gateway mode.

closed 2020-03-29 by marcschier