Azure/Industrial-IoT #53: Make QueueSize configurable

sandervandevelde, 2019-08-03:
We use the OPCPublisher now in production at our customer to read ~16 values from a plc every 50 milliseconds.

We monitor the ingest and everything seems to works fine except for one thing: once every minute (on average) several node values are skipped. We know this because the values produced contain a counter.
 
We traced this back to the QueueSize:

https://github.com/Azure/iot-edge-opc-publisher/blob/master/opcpublisher/OpcMonitoredItem.cs#L577 
 
If we increase this to 50, we retrieve all data as expected.

Please make this QueueSize configurable.

imZack, 2020-01-30:
Any updates on this?

abadar05, 2020-05-18:
https://github.com/Azure/iot-edge-opc-publisher/pull/225#issuecomment-517189424

Any updates on this feature ?

JohannesM24, 2021-04-13:
Any updates on this feature?

dacolgit, 2022-03-31:
in version [2.8.2](https://github.com/Azure/Industrial-IoT/tree/release/2.8.2) QueueSize is configurable for Standalone mode.
For option syntax please look for `DefaultQueueSize` in [publisher-commandline.md](https://github.com/Azure/Industrial-IoT/blob/main/docs/modules/publisher-commandline.md) file.

closed 2022-03-31 by dacolgit