Azure/Industrial-IoT #58: Using ToUniversalTime conversion in HeartBeat message should be ommited

MarekLani, 2019-08-03:
https://github.com/Azure/iot-edge-opc-publisher/blob/a82d885971608f928492060c2e54ffb0de315ba4/opcpublisher/OpcMonitoredItem.cs line 618 contains TimeStamp creation for heartbeat message and there ToUniversalTime conversion used:

`HeartbeatMessage.SourceTimestamp = sourceTimestamp.ToUniversalTime().ToString("o", CultureInfo.InvariantCulture);`   

This is not in pair with "value change" messages/notifications where ToUniversalTime conversion is not used. Please correct me if I am wrong and there is reason for this, but in my opinion this just causes inconsistency in SourceTimeStamp output values. HeartBeat timestamp would be created correctly holding timezone information even without ToUniversalTime conversion.

marcschier, 2019-08-08:
This will be part of 2.5.0 release.

marcschier, 2019-09-07:
Please use 2.5.1 which includes this fix

closed 2019-09-07 by marcschier