Azure/Industrial-IoT #59: OPC Publisher IotHubCommunicationException While Creating ModuleClient

ravindra-gaikawad, 2019-08-03:
I deployed OPC publisher as IotEdge module using [this link](https://github.com/Azure/iot-edge-opc-publisher#using-it-as-a-module-in-azure-iot-edge)

Module fails execution and publisher exists.
Logs are as given below

IoTEdge detected.
[19:49:28 INF] Current directory is: C:\appdata
[19:49:28 INF] Log file is: publisher-publisher.log
[19:49:28 INF] Log level is: info
[19:49:28 INF] OPC Publisher V2.4.0+2 starting up...
[19:49:28 INF] OPC UA monitored item notifications with one of the following 2 status codes will not generate telemetry events:
[19:49:28 INF] StatusCode: BadNoCommunication (dec: 2150694912, hex: 80310000)
[19:49:28 INF] StatusCode: BadWaitingForInitialData (dec: 2150760448, hex: 80320000)
[19:49:28 INF] opcstacktracemask set to: 0x0
[19:49:28 INF] OperationTimeout set to 120000
[19:49:28 INF] OPC UA server base address: opc.tcp://publisher:62222/UA/Publisher
[19:49:28 INF] Security policy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256 with mode SignAndEncrypt added
[19:49:28 INF] Trusted Issuer store type is: Directory
[19:49:28 INF] Trusted Issuer Certificate store path is: pki/issuer
[19:49:28 INF] Trusted Peer Certificate store type is: Directory
[19:49:28 INF] Trusted Peer Certificate store path is: pki/trusted
[19:49:28 INF] Rejected certificate store type is: Directory
[19:49:28 INF] Rejected Certificate store path is: pki/rejected
[19:49:28 INF] Rejection of SHA1 signed certificates is disabled
[19:49:28 INF] Minimum certificate key size set to 1024
[19:49:28 INF] Application Certificate store type is: Directory
[19:49:28 INF] Application Certificate store path is: pki/own
[19:49:28 INF] Application Certificate subject name is: publisher
[19:49:28 WRN] WARNING: Automatically accepting certificates. This is a security risk.
[19:49:28 INF] Application certificate with thumbprint '5D5D27700F3CEDA4D8B27F6A38E76DBC2EE31C79' found in the application certificate store.
[19:49:28 INF] Application certificate is for ApplicationUri 'urn:publisher:publisher:microsoft:', ApplicationName 'publisher' and Subject is 'publisher'
[19:49:28 INF] LDS(-ME) registration intervall set to 0 ms (0 means no registration)
[19:49:28 INF] Trusted issuer store contains 0 certs
[19:49:28 INF] Trusted issuer store has 0 CRLs.
[19:49:28 INF] Trusted peer store contains 0 certs
[19:49:28 INF] Trusted peer store has 0 CRLs.
[19:49:28 INF] Rejected certificate store contains 0 certs
[19:49:28 INF] There is no site configured.
[19:49:28 INF] Starting server on endpoint opc.tcp://publisher:62222/UA/Publisher ...
[19:49:29 INF] Server started.
[19:49:29 INF] Using default telemetry configuration.
[19:49:29 INF] Create module client using 'Amqp_Tcp_Only' for communication.
[19:49:30 INF] IoTCentral mode: False
[19:49:31 ERR] Connection status changed to 'Disconnected', reason 'Retry_Expired'
[19:53:31 ERR] Failure initializing hub communication processing.
Microsoft.Azure.Devices.Client.Exceptions.IotHubCommunicationException: Transient network error occurred, please retry. ---> System.Net.Sockets.SocketException: No such host is known
   at Microsoft.Azure.Devices.Client.Transport.Amqp.AmqpConnector.InitializeTransport(TimeSpan timeout)
   at Microsoft.Azure.Devices.Client.Transport.Amqp.AmqpConnector.OpenConnectionAsync(TimeSpan timeout)
   at Microsoft.Azure.Devices.Client.Transport.Amqp.AmqpConnectionHolder.EnsureConnection(TimeSpan timeout)
   at Microsoft.Azure.Devices.Client.Transport.Amqp.AmqpConnectionHolder.AmqpSessionCreator(DeviceIdentity deviceIdentity, ILinkFactory linkFactory, AmqpSessionSettings amqpSessionSettings, TimeSpan timeout)
   at Microsoft.Azure.Devices.Client.Transport.Amqp.AmqpUnit.OpenAsync(TimeSpan timeout)
   at Microsoft.Azure.Devices.Client.Transport.Amqp.AmqpTransportHandler.OpenAsync(CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.Transport.ProtocolRoutingDelegatingHandler.OpenAsync(CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.Transport.ErrorDelegatingHandler.<>c__DisplayClass22_0.<<ExecuteWithErrorHandlingAsync>b__0>d.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at Microsoft.Azure.Devices.Client.Transport.ErrorDelegatingHandler.ExecuteWithErrorHandlingAsync[T](Func`1 asyncOperation)
   --- End of inner exception stack trace ---
   at Microsoft.Azure.Devices.Client.Transport.ErrorDelegatingHandler.ExecuteWithErrorHandlingAsync[T](Func`1 asyncOperation)
   at Microsoft.Azure.Devices.Client.Transport.RetryDelegatingHandler.<>c__DisplayClass32_0.<<OpenAsyncInternal>b__0>d.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at Microsoft.Azure.Devices.Client.Transport.RetryDelegatingHandler.EnsureOpenedAsync(CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.InternalClient.OpenAsync()
   at OpcPublisher.HubCommunicationBase.InitHubCommunicationAsync(IHubClient hubClient) in C:\app\opcpublisher\HubCommunicationBase.cs:line 202
[19:53:31 FTL] Cannot create module client. Exiting...
[19:53:31 FTL]    at OpcPublisher.IotEdgeHubCommunication..ctor() in C:\app\opcpublisher\IotEdgeHubCommunication.cs:line 43
   at OpcPublisher.IotEdgeHubCommunication.get_Instance() in C:\app\opcpublisher\IotEdgeHubCommunication.cs:line 23
   at OpcPublisher.Program.MainAsync(String[] args) in C:\app\opcpublisher\Program.cs:line 729
System.Exception: Cannot create module client. Exiting...
   at OpcPublisher.IotEdgeHubCommunication..ctor() in C:\app\opcpublisher\IotEdgeHubCommunication.cs:line 43
   at OpcPublisher.IotEdgeHubCommunication.get_Instance() in C:\app\opcpublisher\IotEdgeHubCommunication.cs:line 23
   at OpcPublisher.Program.MainAsync(String[] args) in C:\app\opcpublisher\Program.cs:line 729
[19:53:31 FTL] Publisher exiting...
PS C:\WINDOWS\system32>

Ramji1411, 2019-09-13:
I am facing a similar problem. Can you please confirm if you were able to resolve it and how did you resolve it?

[12:43:33 INF] Current directory is: /app
[12:43:34 INF] Log file is: publisher-publisher.log
[12:43:34 INF] Log level is: info
[12:43:34 INF] OPC Publisher V2.5.1 starting up...
[12:43:34 INF] OPC UA monitored item notifications with one of the following 2 status codes will not generate telemetry events:
[12:43:34 INF] StatusCode: BadNoCommunication (dec: 2150694912, hex: 80310000)
[12:43:34 INF] StatusCode: BadWaitingForInitialData (dec: 2150760448, hex: 80320000)
[12:43:34 INF] opcstacktracemask set to: 0x0
[12:43:34 INF] OperationTimeout set to 120000
[12:43:34 INF] OPC UA server base address: opc.tcp://publisher:62222/UA/Publisher
[12:43:34 INF] Security policy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256 with mode SignAndEncrypt added
[12:43:34 INF] Trusted Issuer store type is: Directory
[12:43:34 INF] Trusted Issuer Certificate store path is: pki/issuer
[12:43:34 INF] Trusted Peer Certificate store type is: Directory
[12:43:34 INF] Trusted Peer Certificate store path is: pki/trusted
[12:43:34 INF] Rejected certificate store type is: Directory
[12:43:34 INF] Rejected Certificate store path is: pki/rejected
[12:43:34 INF] Rejection of SHA1 signed certificates is disabled
[12:43:34 INF] Minimum certificate key size set to 1024
[12:43:34 INF] Application Certificate store type is: Directory
[12:43:34 INF] Application Certificate store path is: pki/own
[12:43:34 INF] Application Certificate subject name is: publisher
[12:43:34 WRN] WARNING: Automatically accepting certificates. This is a security risk.
[12:43:34 INF] No existing Application certificate found. Create a self-signed Application certificate valid from yesterday for 12 months,
[12:43:34 INF] with a 2048 bit key and 256 bit hash.
[12:43:46 INF] Application certificate with thumbprint '6729A7BCEA54530674DC4C6D3861656761D07E13' created.
[12:43:46 INF] Application certificate is for ApplicationUri 'urn:publisher:publisher:microsoft:', ApplicationName 'publisher' and Subject is 'publisher'
[12:43:46 INF] Adding server certificate to trusted peer store. StorePath=pki/trusted
[12:43:46 INF] LDS(-ME) registration intervall set to 0 ms (0 means no registration)
[12:43:46 INF] Trusted issuer store contains 0 certs
[12:43:46 INF] Trusted issuer store has 0 CRLs.
[12:43:46 INF] Trusted peer store contains 1 certs
[12:43:46 INF] 01: Subject 'CN=publisher' (thumbprint: 6729A7BCEA54530674DC4C6D3861656761D07E13)
[12:43:48 INF] Trusted peer store has 0 CRLs.
[12:43:48 INF] Rejected certificate store contains 0 certs
[12:43:48 INF] There is no site configured.
[12:43:48 INF] Starting server on endpoint opc.tcp://publisher:62222/UA/Publisher ...
[12:43:53 INF] Server started.
[12:43:53 INF] Using default telemetry configuration.
[12:43:53 INF] Create module client using 'Amqp_Tcp_Only' for communication.
[12:43:54 INF] IoTCentral mode: False
[12:43:57 ERR] Connection status changed to 'Disconnected', reason 'Retry_Expired'
[12:44:21 INF] Connection status changed to 'Connected', reason 'Connection_Ok'
[12:44:23 INF] Message processing and hub communication configured with a send interval of 10 sec and a message buffer size of 262144 bytes.
[12:44:23 INF] Creating task process and batch monitored item data updates...
[12:44:23 INF] The name of the configuration file for published nodes is: ./pn.json
[12:44:23 INF] The node configuration file './pn.json' does not exist. Continue and wait for remote configuration requests.
[12:44:23 INF] There are 0 nodes to publish.

markusstuhler, 2020-02-20:
Could you try that again with the latest version of the publisher? It seems like a transient network error with IoT Hub. Thank!

Markus

closed 2020-03-09 by marcschier