Azure/Industrial-IoT #60: opc publisher 2.4.1 leaves entries in node configuration file after unpublishing all nodes

anhame, 2019-08-03:
Hello,

when unpublishing all nodes or a list of nodes that will finally result in no results being published, the opcpublishednodes.json still contains one entry.

This is the log snippet:
2019-07-26 17:26:35.830 +02:00 [DBG] HandleUnpublishNodesMethodAsync: called
2019-07-26 17:26:35.832 +02:00 [INF] HandleUnpublishNodesMethodAsync: Request to stop monitoring item with NodeId 'ns=5;s=PumpStatus1')
2019-07-26 17:26:35.834 +02:00 [INF] RequestMonitorItemRemoval: Node with id 'ns=5;s=PumpStatus1' tagged to stop monitoring.
2019-07-26 17:26:35.835 +02:00 [DBG] HandleUnpublishNodesMethodAsync: Id 'ns=5;s=PumpStatus1': tagged for removal
2019-07-26 17:26:35.837 +02:00 [INF] HandleUnpublishNodesMethodAsync: Request to stop monitoring item with NodeId 'ns=5;s=Temperature')
2019-07-26 17:26:35.839 +02:00 [INF] RequestMonitorItemRemoval: Node with id 'ns=5;s=Temperature' tagged to stop monitoring.
2019-07-26 17:26:35.840 +02:00 [DBG] HandleUnpublishNodesMethodAsync: Id 'ns=5;s=Temperature': tagged for removal
2019-07-26 17:26:35.840 +02:00 [INF] Remove nodes in subscription with id 19 on endpoint 'opc.tcp://xxx'
2019-07-26 17:26:35.841 +02:00 [INF] HandleUnpublishNodesMethodAsync: Request to stop monitoring item with NodeId 'ns=5;s=VentilationLevel1')
2019-07-26 17:26:35.842 +02:00 [INF] There are now 1 monitored items in this subscription.
2019-07-26 17:26:35.844 +02:00 [INF] There are now 1 items managed by publisher for this subscription. (version: 00000009)
2019-07-26 17:26:35.845 +02:00 [INF] RequestMonitorItemRemoval: Node with id 'ns=5;s=VentilationLevel1' tagged to stop monitoring.
2019-07-26 17:26:35.847 +02:00 [DBG] HandleUnpublishNodesMethodAsync: Id 'ns=5;s=VentilationLevel1': tagged for removal
2019-07-26 17:26:35.848 +02:00 [INF] HandleUnpublishNodesMethodAsync: completed with result OK
2019-07-26 17:26:35.858 +02:00 [INF] Remove nodes in subscription with id 19 on endpoint 'opc.tcp://xxx'
2019-07-26 17:26:35.861 +02:00 [INF] There are now 0 monitored items in this subscription.
2019-07-26 17:26:35.862 +02:00 [INF] There are now 0 items managed by publisher for this subscription. (version: 0000000A)
2019-07-26 17:26:35.863 +02:00 [INF] Remove unused subscriptions on endpoint 'opc.tcp://xxx'.
2019-07-26 17:26:35.872 +02:00 [INF] There are now 0 subscriptions in this session.
2019-07-26 17:26:35.873 +02:00 [DBG] OPC: ***EXCEPTION*** BadNoSubscription
2019-07-26 17:26:35.874 +02:00 [INF] Remove unused session on endpoint 'opc.tcp://xxx'.
2019-07-26 17:26:35.877 +02:00 [INF] Removing 0 subscriptions from session.
2019-07-26 17:26:35.878 +02:00 [INF] Closing session to endpoint URI 'opc.tcp://sba020163.sgr.com:53530/OPCUA/SimulationServer' closed successfully.
2019-07-26 17:26:35.899 +02:00 [INF] Session to endpoint URI 'opc.tcp://xxx' closed successfully.

But publishednodes.json still has one entry left for the corresponding endpointUrl. It is the last one that has been removed.

best regards
Hartmut

anhame, 2019-08-08:
This issue has already been solved by my pull request in the publisher repo.
It was merged in commit be4bf5c6ae6f5be23a9e1edbbb2cf06b2a9ff7e0
"move final changes to publisher from publisher repo"

marcschier, 2019-08-08:
Thank you @anhame, I moved this over to align with a new release, but it looks like it was already part of 2.4.1 then.  I will close this then.

closed 2019-08-08 by marcschier