Azure/Industrial-IoT #69: GIT dependency in run.ps1

BinEilig, 2019-08-07:
**Describe the bug**
the deploy script will start the 'vm\run.ps1' script, which calls the 'git' command (line 227).
If git is not in the search path of the machine, the deployment will fail.

**To Reproduce**
start the deployment without git in the search path.

**Expected behavior**
a) advice in the notes, that git is a prerequisite for the deployment
b) check for git command in the script

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Win10, Powershell
 - Browser [e.g. chrome, safari]
 - Version Win10

marcschier, 2019-08-08:
Thank you @BinEilig - Good catch - this should be fixed by not requiring git.

marcschier, 2019-08-16:
In 2.5.0

closed 2019-08-16 by marcschier