Azure/Industrial-IoT #81: Read from several OPC UA servers

AlexGalvarez, 2019-08-19:
Hi!

How do I configure the publisher module to read from **several OPC UA servers**?

I have two endpoint URLs (two OPC UA servers) and I don't know how to configure the pn.json file to include both data sources.

It seems to me that is something not very complicated but I don't have the solution yet.

Many thanks!

ryan-ms, 2019-08-20:
Hi @AlexGalvarez ,

Try this

`[
  {
    "EndpointUrl": "opc.tcp://192.168.4.104:4840/",
    "UseSecurity": false,
    "OpcNodes": [
      {
        "Id": "nsu=TagProvider;s=opctag1"
      },
      {
        "Id": "nsu=TagProvider;s=opctag2"
      }
    ]
  },
  {
    "EndpointUrl": "opc.tcp://192.168.4.105:4840/",
    "UseSecurity": false,
    "OpcNodes": [
      {
        "Id": "nsu=TagProvider;s=opctag3"
      },
      {
        "Id": "nsu=TagProvider;s=opctag4"
      }
    ]
  }
]`

AlexGalvarez, 2019-08-20:
Thank you very much @ryan-ms !

It works properly.

closed 2019-08-20 by AlexGalvarez