Azure/Industrial-IoT #84: Missing Azure AD permissions

eXXL, 2019-08-26:
**Describe the bug**
I don't have permissions to create the app registrations in Azure AD. It is managed by a global admin team.

**Expected behavior**
I would like to pre-create the app registrations and instruct the script to use the existing app registrations so no admin permissions to Azure AD is required to run the deployment script.

mma-mm, 2019-08-26:
This is also a problem for me.

marcschier, 2019-09-07:
We will address this in a future release where we plan to write a full deployment tool in C#. Likely eta is end of October.

marcschier, 2020-02-19:
We added this functionality to the deploy.ps1 script.  You can pre-create the aad registrations using aad-register.ps1 and then pass the output file to deploy.ps1 as parameter to use.

closed 2020-02-19 by marcschier