Azure/Industrial-IoT #85: How to use IBrowseServices with EndpointRegistrationModel with Username/Password authentication?

mma-mm, 2019-08-29:
Hi, I'd like to use the I BrowseServices with an OPC Server that only allows Authentication with Username + Password and SecurityPolicy and SecurityMode "None". How do i need to configure the EndpointRegistrationModel and the EndpointModel?

marcschier, 2019-09-04:
Hi @mma-mm, that should technically be possible.   

The EndpointModel contains security profile and security mode settings which you can set to None.

For credentials, each request model has a RequestHeader header, which has an Elevation property, which takes a credential model.  The credential in there is a JToken, but https://github.com/Azure/Industrial-IoT/blob/d7fec955abb8e89e91736e321124b16dffa076be/components/opc-ua/src/Microsoft.Azure.IIoT.OpcUa.Protocol/src/Extensions/StackModelsEx.cs#L341 shows how to formulate this json for e.g. user name password authentication.   The properties are "user" and "password", i.e. 

```
                        var value = JToken.FromObject(new {
                            user = un.UserName,
                            password = un.DecryptedPassword
                        })
```

I would however suggest that sending plain passwords over security none connections means between the twin module/browseservice and the server the password is sent as clear text.   This should absolutely be avoided in production scenarios.

Hope this help.

Marc

marcschier, 2020-03-19:
Closing this - we will have a better way to use authentication as per other tracking issues.

closed 2020-03-19 by marcschier