Azure/Industrial-IoT #88: Microsoft.Azure.ServiceBus.ServiceBusCommunicationException Using Discover

sjai013, 2019-09-06:
I get a Microsoft.Azure.ServiceBus.ServiceBusCommunicationException exception when trying to use the Discover API of the OPC Registry (via Swagger) - I am running all services via docker, using the most current master branch.  

Any help would be appreciated.  Please let me know what other information would be useful.

`
Exception has occurred: CLR/Microsoft.Azure.ServiceBus.ServiceBusCommunicationException
Exception thrown: 'Microsoft.Azure.ServiceBus.ServiceBusCommunicationException' in System.Private.CoreLib.dll: 'A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond ErrorCode: TimedOut'
 Inner exceptions found, see $exception in variables window for more details.
 Innermost exception 	 System.Net.Sockets.SocketException : A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
...
`

marcschier, 2019-09-07:
Please redeploy as the platform now depends on service is which wasn't previously deployed.  If you see a service bus connection string in your .env file, please ensure that the jobs service is running.  We will in parallel double check this scenario, although it in particular is part of our acceptance tests before release.

sjai013, 2019-09-09:
Thanks, I'm not getting this exception anymore.

closed 2019-09-09 by sjai013