Azure/Industrial-IoT #94: When calling NodePublishListAsync in PublisherServices, NodeId does not contain namespace

mma-mm, 2019-09-13:
**Describe the bug**
When NodePublishListAsync in PublisherServices is called, the Node id of the returned nodes does not contain any namespace identifier. The node was published using a namespace identifier before.

**To Reproduce**
Steps to reproduce the behavior:
1. Deploy publisher without publishednodes.json
2. Use PublisherServices
3. Publish a Node with a nodeId like this: "ns=1;s=path/to/node"
4. Call NodePublishListAsync
5. returned NodeId is "s=path/to/node"

**Expected behavior**
The returned node id contains the namespace in any way.

**Screenshots**

**Desktop (please complete the following information):**
Ubuntu 18.04 x64
IoTEdge 1.0.9RC1
Publisher:  2.3.3 and latest

marcschier, 2019-09-26:
Hi @mma-mm - the reason this does not work today is that the publish command uses the node id string it is passed as-is when passing to publisher - in this case with node id index (ns=1).   Publisher finds the index in the NS table and without conversion stores the passed node id string in the publishednodes.json.   When the original string is returned to the publisherservices.cs during listing, the twin module tries to convert the returned id string to a node using its global namespace table which does not contain ns=1 (or the table has nothing meaningful).   And thus the ns=1 is stripped off.   

What we could do is
- Check whether the returned string has a ns= and if so, return the original node id string as is.   This is a band aid though, but I believe would make your scenario work.  
- In addition - convert a ns!=0 on the publisher side into a namespace uri and save the id in nsu=uri form.   This would be a more correct fix since the index could change between publisher sessions.  

I am going to try the first approach as it is simple and given that we are doing larger changes in the publisher at the moment probably more expeditious.

marcschier, 2019-12-02:
Fixed in 2.5.2

closed 2019-12-02 by marcschier