Azure/Industrial-IoT #103: IoT Hub Connection Issue

YashC, 2019-09-15:
**Describe the bug**
Failing to send telemetry from device to cloud with connection issues with IoT Edge setup and OPC Twin and Publisher module. 

Following is the error message and call stack for the error from edge hub. 

Finished processing messages to upstream
2019-09-15T22:59:03.834165700Z <7> 2019-09-16 06:59:03.834 +08:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.CloudProxy.ConnectivityAwareClient] - Received connection status changed callback with connection status Disconnected_Retrying and reason Communication_Error for BMSOPC/$edgeHub
2019-09-15T22:59:03.839166600Z <7> 2019-09-16 06:59:03.839 +08:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.CloudProxy.ConnectivityAwareClient] - Received connection status changed callback with connection status Disconnected_Retrying and reason Communication_Error for BMSOPC/opcpublisher
2019-09-15T22:59:05.362127700Z <7> 2019-09-16 06:59:05.362 +08:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.CloudProxy.ConnectivityAwareClient] - Received connection status changed callback with connection status Connected and reason Connection_Ok for BMSOPC/$edgeHub
2019-09-15T22:59:05.381119700Z <7> 2019-09-16 06:59:05.381 +08:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.CloudProxy.ConnectivityAwareClient] - Received connection status changed callback with connection status Connected and reason Connection_Ok for BMSOPC/opcpublisher
2019-09-15T22:59:05.434113000Z <7> 2019-09-16 06:59:05.434 +08:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.CloudProxy.ConnectivityAwareClient] - Received connection status changed callback with connection status Disconnected_Retrying and reason Communication_Error for BMSOPC/opcpublisher
2019-09-15T22:59:05.435110900Z <7> 2019-09-16 06:59:05.435 +08:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.CloudProxy.ConnectivityAwareClient] - Received connection status changed callback with connection status Disconnected_Retrying and reason Communication_Error for BMSOPC/$edgeHub
2019-09-15T22:59:06.943508300Z <7> 2019-09-16 06:59:06.943 +08:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.CloudProxy.ConnectivityAwareClient] - Received connection status changed callback with connection status Connected and reason Connection_Ok for BMSOPC/$edgeHub
2019-09-15T22:59:06.943508300Z <7> 2019-09-16 06:59:06.943 +08:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.CloudProxy.ConnectivityAwareClient] - Received connection status changed callback with connection status Connected and reason Connection_Ok for BMSOPC/opcpublisher
2019-09-15T22:59:06.992809300Z <7> 2019-09-16 06:59:06.992 +08:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.CloudProxy.ConnectivityAwareClient] - Received connection status changed callback with connection status Disconnected_Retrying and reason Communication_Error for BMSOPC/opcpublisher
2019-09-15T22:59:07.006801700Z <7> 2019-09-16 06:59:07.007 +08:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.CloudProxy.ConnectivityAwareClient] - Received connection status changed callback with connection status Disconnected_Retrying and reason Communication_Error for BMSOPC/$edgeHub
2019-09-15T22:59:08.531031700Z <7> 2019-09-16 06:59:08.531 +08:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.CloudProxy.ConnectivityAwareClient] - Received connection status changed callback with connection status Connected and reason Connection_Ok for BMSOPC/$edgeHub
2019-09-15T22:59:08.531031700Z <7> 2019-09-16 06:59:08.531 +08:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.CloudProxy.ConnectivityAwareClient] - Received connection status changed callback with connection status Connected and reason Connection_Ok for BMSOPC/opcpublisher
2019-09-15T22:59:08.582233000Z <7> 2019-09-16 06:59:08.582 +08:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.CloudProxy.ConnectivityAwareClient] - Received connection status changed callback with connection status Disconnected_Retrying and reason Communication_Error for BMSOPC/opcpublisher
2019-09-15T22:59:08.582233000Z <7> 2019-09-16 06:59:08.582 +08:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.CloudProxy.ConnectivityAwareClient] - Received connection status changed callback with connection status Disconnected_Retrying and reason Communication_Error for BMSOPC/$edgeHub
2019-09-15T22:59:10.036943500Z <7> 2019-09-16 06:59:10.036 +08:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.CloudProxy.ConnectivityAwareClient] - Received connection status changed callback with connection status Connected and reason Connection_Ok for BMSOPC/$edgeHub
2019-09-15T22:59:10.060916000Z <7> 2019-09-16 06:59:10.061 +08:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.CloudProxy.ConnectivityAwareClient] - Received connection status changed callback with connection status Connected and reason Connection_Ok for BMSOPC/opcpublisher
2019-09-15T22:59:10.119151900Z <7> 2019-09-16 06:59:10.119 +08:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.CloudProxy.ConnectivityAwareClient] - Received connection status changed callback with connection status Disconnected_Retrying and reason Communication_Error for BMSOPC/$edgeHub
2019-09-15T22:59:10.142150900Z <7> 2019-09-16 06:59:10.142 +08:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.CloudProxy.ConnectivityAwareClient] - Received connection status changed callback with connection status Disconnected_Retrying and reason Communication_Error for BMSOPC/opcpublisher
2019-09-15T22:59:11.563516000Z <7> 2019-09-16 06:59:11.564 +08:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.CloudProxy.ConnectivityAwareClient] - Received connection status changed callback with connection status Connected and reason Connection_Ok for BMSOPC/opcpublisher
2019-09-15T22:59:11.580636800Z <7> 2019-09-16 06:59:11.581 +08:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.CloudProxy.ConnectivityAwareClient] - Received connection status changed callback with connection status Connected and reason Connection_Ok for BMSOPC/$edgeHub
2019-09-15T22:59:11.622884900Z <7> 2019-09-16 06:59:11.623 +08:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.CloudProxy.ConnectivityAwareClient] - Received connection status changed callback with connection status Disconnected_Retrying and reason Communication_Error for BMSOPC/$edgeHub
2019-09-15T22:59:11.641890500Z <7> 2019-09-16 06:59:11.642 +08:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.CloudProxy.ConnectivityAwareClient] - Received connection status changed callback with connection status Disconnected_Retrying and reason Communication_Error for BMSOPC/opcpublisher
2019-09-15T22:59:13.097188400Z <7> 2019-09-16 06:59:13.097 +08:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.CloudProxy.ConnectivityAwareClient] - Received connection status changed callback with connection status Connected and reason Connection_Ok for BMSOPC/$edgeHub
2019-09-15T22:59:13.121182000Z <7> 2019-09-16 06:59:13.121 +08:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.CloudProxy.ConnectivityAwareClient] - Received connection status changed callback with connection status Connected and reason Connection_Ok for BMSOPC/opcpublisher
2019-09-15T22:59:15.433417500Z <7> 2019-09-16 06:59:15.434 +08:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.CloudProxy.ConnectivityAwareClient] - Received connection status changed callback with connection status Disconnected_Retrying and reason Communication_Error for BMSOPC/$edgeHub
2019-09-15T22:59:15.451426500Z <7> 2019-09-16 06:59:15.451 +08:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.CloudProxy.ConnectivityAwareClient] - Received connection status changed callback with connection status Disconnected_Retrying and reason Communication_Error for BMSOPC/opcpublisher
2019-09-15T22:59:16.901099900Z <7> 2019-09-16 06:59:16.901 +08:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.CloudProxy.ConnectivityAwareClient] - Received connection status changed callback with connection status Connected and reason Connection_Ok for BMSOPC/opcpublisher
2019-09-15T22:59:16.921325600Z <7> 2019-09-16 06:59:16.922 +08:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.CloudProxy.ConnectivityAwareClient] - Received connection status changed callback with connection status Connected and reason Connection_Ok for BMSOPC/$edgeHub
2019-09-15T22:59:22.918122800Z <7> 2019-09-16 06:59:22.918 +08:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.CloudProxy.ConnectivityAwareClient] - Received connection status changed callback with connection status Disconnected_Retrying and reason Communication_Error for BMSOPC/$edgeHub
[... cut ...] ivityAwareClient.cs:line 134
2019-09-15T22:59:32.970481200Z <7> 2019-09-16 06:59:32.969 +08:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.CloudProxy.CloudProxy] - Error sending message for device BMSOPC/opcpublisher in cloud proxy 91c5baed-b680-4a2e-bb54-6e314dd03b22

Full output file attached here
[edgeHub.log](https://github.com/Azure/Industrial-IoT/files/3614410/edgeHub.log)

** System Config:**
 - OS: Windows 10 Enterprise LTSC
 - Hardware: 64bit,  8 GB RAM, 
 - Internet: 4G LTE (wireless card)

YashC, 2019-09-16:
I also tried running the OPC publisher directly as a docker container using instructions over here
[ Using a container from Microsoft Container Registry ](https://github.com/Azure/Industrial-IoT/blob/master/docs/modules/publisher.md#using-a-container-from-microsoft-container-registry)

It produced similar output with continuous loop of error messages like 

> Received connection status changed callback with connection status Disconnected_Retrying and reason Communication_Error
>  Received connection status changed callback with connection status Connected and reason Connection_Ok

Options used for the command
```
docker -H npipe:////./pipe/iotedge_moby_engine `
       run --name opcpublisher  -p 62222:62222 -v c:/IoTEdgeMapping:c:/publisher --network host --ip 192.168.1.110  -h publisher mcr.microsoft.com/iotedge/opc-publisher:2.5.0-windows-amd64 `
       publisher -c $iothubConString --dc $deviceConString --pf /publisher/pn.json --aa --di 60 --to --si 10 --ms 262144 --mq 32768 --oi 30000 --op 60000 
```
Edge setup in  the previous message and above command works fine on a VM with window IOT enterprise but fails on a Intel NUC with 4G Wifi card. 

Any help to diagnose the issue would be really helpful. Thanks.

marcschier, 2019-09-17:
Have you tried a different hardware or used wired adapter?  I am guessing but this looks like packet loss or corruption causing disconnects.

marcschier, 2019-09-19:
Another reason for many reconnects is if the same connection string is used from 2 different publishers or edges.  Another item to check.   Also, the edge has a connectivity check.  You can run it using iotedge check, it also provides useful diagnostics.

YashC, 2019-09-19:
1. We haven't tried switching to different hardware but in the coming days, we will be trying a wired connection. 
2. Currently, we just have one edge device and one publisher running in it.   
3. 'iotedge check' returns fine
```
PS C:\> iotedge check
Configuration checks
--------------------
√ config.yaml is well-formed - OK
√ config.yaml has well-formed connection string - OK
√ container engine is installed and functional - OK
√ Windows host version is supported - OK
√ config.yaml has correct hostname - OK
√ config.yaml has correct URIs for daemon mgmt endpoint - OK
√ latest security daemon - OK
√ host time is close to real time - OK
√ container time is close to host time - OK
√ DNS server - OK
‼ production readiness: certificates - Warning
    Device is using self-signed, automatically generated certs.
    Please see https://aka.ms/iotedge-prod-checklist-certs for best practices.
√ production readiness: certificates expiry - OK
√ production readiness: container engine - OK
√ production readiness: logs policy - OK

Connectivity checks
-------------------
√ host can connect to and perform TLS handshake with IoT Hub AMQP port - OK
√ host can connect to and perform TLS handshake with IoT Hub HTTPS / WebSockets port - OK
√ host can connect to and perform TLS handshake with IoT Hub MQTT port - OK
√ container on the IoT Edge module network can connect to IoT Hub AMQP port - OK
√ container on the IoT Edge module network can connect to IoT Hub HTTPS / WebSockets port - OK
√ container on the IoT Edge module network can connect to IoT Hub MQTT port - OK
√ Edge Hub can bind to ports on host - OK

20 check(s) succeeded.
1 check(s) raised warnings. Re-run with --verbose for more details.
```

markusstuhler, 2019-11-18:
Does the issue still persist, also with a different network adapter?
I will close it for now due to inactivity. Please reopen if needed.
Thanks!

closed 2019-11-18 by markusstuhler