Azure/Industrial-IoT #104: OPC Twin - Direct Messaging Info

sjai013, 2019-09-16:
I'm trying to test writing to OPC tags using OPC Twin, but I don't want to have to deploy all the services that go with the deploy option (because I just want to test).

Is there documentation somewhere on using Direct Messaging on the OPC Twin to read/write OPC tags, or alternatively hosting the microservices completely locally (i.e. no deployment to Azure, other than that required for OPC Publisher and OPC Twin, and therefore very few parameters in the .env file actually relevant)?

marcschier, 2019-09-30:
See #85

marcschier, 2020-02-19:
We recently added API surface in our API packages for the twin that allows calling direct method directly.  Since we are not making guarantees around stability of the Twin module api surface, the version of API and the twin have to match.   My recommendation is to use the REST interface.   We have a long term plan to run the entire platform locally (in iot edge), in which case this would be future proof.

closed 2020-02-19 by marcschier