Azure/Industrial-IoT #106: Allow Application Discovery for a specific supervisor/opc-twin-module

mma-mm, 2019-09-18:
**Is your feature request related to a problem? Please describe.**
Discovering OPC-Servers triggers a discovery on every opc-twin-module on any device connected with the IIoT-Solution.

**Describe the solution you'd like**
I want to be able to trigger a discovery for a specific edge-device/opc-twin-module, as the discovery settings for different device may require different settings.

**Describe alternatives you've considered**
Manually adding the server would be possible, but require a custom implementation of the discovery.

marcschier, 2019-09-19:
Hi @mma-mm, if I understand correctly you are asking for the ability to trigger a one-time discover on a module.   This should be possible as the multiplexed discovery is essentially a single trigger on every module.   We do not have a REST API for it yet, but it should be straightforward to add.

cristipogacean, 2019-09-19:
Hi @mma-mm, try the supervisor update (PATCH /v2/supervisors/{supervisorId}) Rest API, it should allow discovery per module with different discovery settings.

mma-mm, 2019-09-19:
Both sounds good, thank you.

marcschier, 2020-03-02:
Please reopen if the patch does not work for you.  Thanks.

closed 2021-05-11 by hansgschossmann