Azure/Industrial-IoT #107: status of TODO item in parameter credentials from deploy.ps1

sharoot, 2019-09-18:
What is the status of the deploy.ps1 file? 
The credential parameter seems to be a TODO item:
 
.PARAMETER credentials
    To support non interactive usage of script. (TODO)

Is it currently possible to deploy the opc microservices whithout having administrator permissions in your azure account? If not, could you make this possible?

marcschier, 2019-09-19:
Hi @sharoot , We are working on a new deploy experience.   This is great input and we will see how we can accommodate this while the new experience is in development.

sharoot, 2019-09-19:
Hello @marcschier , thank you very much for your reply. Great to hear that you are working on a new deploy experience. When can we expect this new experience?

karok2m, 2019-09-25:
Hi @sharoot, likely ETA of new deployment experience is end of October.

mma-mm, 2019-11-08:
Hi, its november now. What is the current state about this feature?

karok2m, 2020-01-14:
Hi @sharoot and @mma-mm. New deployment is done via `Microsoft.Azure.IIoT.Deployment` application. Its codes can be found in [`deploy/src/Microsoft.Azure.IIoT.Deployment/`](https://github.com/Azure/Industrial-IoT/tree/master/deploy/src/Microsoft.Azure.IIoT.Deployment) directory. It now supports execution using either user or service principal credentials. To use it non-interactively you would have to use service principal credentials authentication flow. You would need to provide credentials and the rest of configuration options via either [appsettings.json](https://github.com/Azure/Industrial-IoT/blob/master/deploy/src/Microsoft.Azure.IIoT.Deployment/appsettings.json) file, command-line or environment variables. Information about authentication flows and configuration parameters can be found in [Microsoft.Azure.IIoT.Deployment documentation](https://github.com/Azure/Industrial-IoT/blob/master/docs/industrial_iot_deployment.md).

Please note that `Microsoft.Azure.IIoT.Deployment` creates whole Azure infrastructure required to run our microservices. And part of that is registration of AAD applications which requires administrator permissions. It is possible to separate the deployment into two stages (via `RunMode` parameter), where first one only performs AAD applicatoin registration and setup, and second one creates Azure resources and deploys microservices. In this case, administrator permissions are required only for the first stage, but Contributor's role would be enough to run the second stage.

marcschier, 2020-01-23:
In addition to the new deployment tool (to deploy into aks) there is still a deploy.ps1 script based deployment, but it moved into /deploy/scripts in master.   

This script now takes a Az PS context object as parameter created e.g. ahead of time using service principal or some other way.   

However, please note that only the new tooling mentioned above is aimed to be used for production deployments within Kubernetes ("latest" tag, currently 2.5.2), the script on the other hand supports quick preview (= state of master branch - currently 2.6.37) and factory simulation deployments whereby all micro services are hosted in the same all in one container in app service now.   I.e. production deployment is always a bit behind.

closed 2020-01-23 by marcschier