Azure/Industrial-IoT #110: Cannot expand the OPC node in some OPC UA server

ryan-ms, 2019-09-19:
**Describe the bug**

Cannot successfully expand the OPC node in some OPC UA server while other OPC UA client can successful expand it.

When the problem occurred, the response from OPC Twin service would show error message: "`BadNodeIdUnknown`". 

**To Reproduce**
Steps to reproduce the behavior:
1.  Runniung OPC Twin and OPC Publisher module in IoT Edge in a ubuntu gateway.
2.  Use OPC Twin webUI to discover, browse OPC UA server.
3.  Using the UI to expand the OPC UA node, then the issue occurred.  Like picture below.

Not showing any OPC tag in Tags node.
[image]

But can see the OPC tag in Tags node while using OPC Client like UAExperts.
[image]

**Expected behavior**
Can successfully expand the OPC node in any OPC UA server just like other OPC client do.

**Screenshots**

Response of expanding the node with  abnormal OPC UA server
[image]

Response of expanding the node with  normal OPC UA server
[image]

**Additional context**
The issue behavior is similar with issue #26  , but I think it's different issue. I've used latest industry-service and OPC Twin module to test. and I think issue #26 is resolved now in my testing environment. But the issue which I reported still happened.

marcschier, 2019-09-19:
Hi @ryan-ms, this looks like a bug resulting from probably double url escaping the namespace uri - once from the GET parameters, and later in the node id parsing.   I will look into it.  

In the meantime could you try the same with the CLI?  Which I believe uses the POST version of browse.

ryan-ms, 2019-09-19:
Hi @marcschier ,

Thanks for reply.

Sure, I can try the CLI. But actually, I am not very familiar to use CLI, I've referenced the [document](https://github.com/Azure/Industrial-IoT/blob/master/modules/opc-twin/cli/Program.cs) to use CLI.

I can used --list ,  --host command which document mentioned , but not sure what command should I use to browse the OPC node.

Could you please provide me the document or guide me how to use CLI to browse node?
Thank you.

By the way, I've tried use POST method to browse node by using postman, but the result is the same.
[image]

marcschier, 2019-09-27:
Hi @ryan-ms, the issue is that the namespace is not a uri, but just "TagProvider".  I will add a unit test to see that we support this case and whether this is an issue in the GET or sample client.  In the meantime, instead of the above can you use "nsu=TagProvider;s=Tags" as node id using POST in postman?

ryan-ms, 2019-10-02:
Hi @marcschier ,

Yes, I've tried "nsu=TagProvider;s=Tags" as node id using POST in postman.
And the response is good!
[image]

So I assume the Connected Factory or OPC Twin client will update to use POST method in the future, right?!  

Thanks for your response.

marcschier, 2019-10-02:
Hi @ryan-ms, yes, but I also now have checked in unit tests in master that exercise both POST nad the GET path wtih a bad URI, and they indicated that everything is working ok (I did not need to make any real fixes).  

Connected factory uses the API nuget which is doing POST, but older services, so I assume that this is the issue there and we have fixed something since 2.0.1.   For the webapp, my assumption is that it is doing something wrong such as not url encoding the node id.

closed 2019-10-02 by marcschier