Azure/Industrial-IoT #111: Building on Mac OS doesn't work

toddysm, 2019-09-19:
**Describe the bug**
We are trying to build the Publisher module on a Mac but it fails with tons of error. Also, we are noticing missing files in the project (particularly Dockerfile) - assuming those are part of the build process. 

Feedback: Documentation can be clearer about how to do the build and create the Docker container.

**To Reproduce**
Steps to reproduce the behavior:
1. Clone the project
2. Open the Solution in VS for Mac
3. Rebuild All
4. See error

**Expected behavior**
The solution to build and to have Dockerfile to create the IoT Edge module.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Desktop (please complete the following information):**
 - OS: Mac OS X 10.14.6
 - Visual Studio for Mac Community 8.2.6 (build 26) [e.g. chrome, safari]
 - Dotnet Core Version 2.2.300

**Additional context**
Add any other context about the problem here.

marcschier, 2019-09-23:
Thanks – Indeed we regressed when it comes to being able to build containers for our modules.   This shall be fixed.   

We however do build and test on macos as part of our CI pipeline, so the build issue might be a tool or setup issue.   Can you try building using "dotnet build Industrial-IoT.sln" in a terminal?

toddysm, 2019-09-23:
@marcschier Building manually from the console works (sorry, forgot to mention that). Is there any particular documentation on how to set up VS on Mac to do the build from there? 

Also, any documentation about the Docker container creation will be appreciated.

marcschier, 2019-09-27:
Hi @toddysm, there should not be anything special around setting up VS for Mac beyond the official documentation.  I would suggest trying the latest 8.3 build.   What looks iffy is the errors relating to mono, maybe this is a path issue.  Unfortunately I do not have a Mac on hand and would not be able to help troubleshoot.   

Regarding publisher, yes it does not require any micro services deployed.  Publisher.md shows how to configure the publishednodes.json manually.   It can also be run outside of Edge and thus directly from command line (also documented).   There will be a build script in the tree soon that allows building the module containers.   I will leave this open to also track that.

vjrantal, 2019-10-31:
While this may not solve the original problem, I wanted to share that I was able to run "Build All" after the opening the solution on VS for Mac version 8.3.1 (build 18).

closed 2020-03-02 by marcschier