Azure/Industrial-IoT #113: Error Azure Account in deployment script

Distil62, 2019-09-23:
**Describe the bug**
Add-AzureAccount throw error `No subscriptions are associated with the logged in account in Azure Service Management (RDFE)`

**To Reproduce**
Steps to reproduce the behavior:
1. clone the project
2. execute the commande `deploy.cmd` without any args
3. log-in
4. see error

**Expected behavior**
I guess the application have to detect my subscription

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - Windows 10 build 18362
 - Google Chrome 76

**Additional context**
I promote this account recently.

Distil62, 2019-09-23:
I read it's a comptability error between the old and new Azure portal but i don't know how can i fix it :/

Distil62, 2019-09-24:
I don't understand what the are required more :/

karok2m, 2019-09-25:
@Distil62, can you please verify that you have admin permissions in your Azure account ? Currently, `deploy.cmd` requires user to be an admin in Azure account.

marcschier, 2019-09-29:
Hi @Distil62, unfortunately we do not support deployment as user at this point.  Please have an administrator deploy.  We are currently reworking the deployment experience and will include deploying as user as a future feature.

marcschier, 2020-02-19:
We implemented a way for a aad admin to do the AAD application registration (aad-register.ps1) and provide the created output as file to the deployment script.   We removed Add-AzureAccount, so this should also not happen anymore.   Please try again and let us know if you still hit issues.

closed 2020-02-19 by marcschier