Azure/Industrial-IoT #115: Heartbeat does not reset when value changes

sjai013, 2019-09-24:
**Describe the bug**
If a heartbeat time has been configured for a tag, the tag data is published every heartbeat, regardless of if it was published due to publishing interval already.

**To Reproduce**
Steps to reproduce the behavior:
1. Set up tag with publishing interval and heartbeat interval (heartbeat interval slightly larger than pubishing interval).
2. Monitor data, and observe data being submitted twice.

**Expected behavior**
Heartbeat should reset once data has been published.

**Additional context**
Using the latest version of OPCPublisher on Azure Marketplace, running on Ubuntu 18.04 Server.

sjai013, 2019-09-24:
As I type this, I am unable to consistently reproduce the bug.  My --oi=1000, --op=600000, --hb=1200.

The following data is returned:
`[{"NodeId":"Counter1","ApplicationUri":"urn:LP416:OPCUA:SimulationServer","DisplayName":"Counter1","Value":1,"SourceTimestamp":"2019-09-24T05:00:01.0000000+00:00"},{"NodeId":"Counter1","ApplicationUri":"urn:LP416:OPCUA:SimulationServer","DisplayName":"Counter1","Value":0,"SourceTimestamp":"2019-09-24T05:00:00.0000000+00:00"}]`

Note the SourceTimestamp difference between the two rows, and the value has also changed (0->1).

markusstuhler, 2019-11-19:
Hi @sjai013 ,
we're currently working on an updated implementation for the heartbeat mechanism which will be more powerfull and rely on OPC UA server capabilities to trigger the messages. We track the progress here: https://github.com/Azure/Industrial-IoT/issues/135

Therefore, I will close this issue.

Thanks!

closed 2019-11-19 by markusstuhler