Azure/Industrial-IoT #116: Container runtime error when starting publisher container

ddobric, 2019-09-24:
## Expected Behavior
edgeAgent throws when trying to start the publisher module.

> Container runtime error (See below pasted log)

It might be that some config was not properly set. Unfortunately, I do not see way to figure out what is exactly the problem. 

## Current Behavior
Module cannot be loaded, because of an error. It is probably some error in module itself or some configuration error.
This is exactly the issue, because it is not know what kind of error it is.

## Steps to Reproduce
Deploy module

## Context (Environment)

### Output of `iotedge check`

<details>
<summary>Click here</summary>

```
Configuration checks
--------------------
√ config.yaml is well-formed - OK
√ config.yaml has well-formed connection string - OK
√ container engine is installed and functional - OK
‼ config.yaml has correct hostname - Warning
    config.yaml has hostname Abcde-SR1 which does not comply with RFC 1035.

    - Hostname must be between 1 and 255 octets inclusive.
    - Each label in the hostname (component separated by ".") must be between 1 and 63 octets inclusive.
    - Each label must start with an ASCII alphabet character (a-z), end with an ASCII alphanumeric character (a-z, 0-9), and must contain only ASCII alphanumeric characters or hyphens (a-z, 0-9, "-").

    Not complying with RFC 1035 may cause errors during the TLS handshake with modules and downstream devices.
√ config.yaml has correct URIs for daemon mgmt endpoint - OK
√ latest security daemon - OK
√ host time is close to real time - OK
√ container time is close to host time - OK
‼ DNS server - Warning
    Container engine is not configured with DNS server setting, which may impact connectivity to IoT Hub.
    Please see https://aka.ms/iotedge-prod-checklist-dns for best practices.
    You can ignore this warning if you are setting DNS server per module in the Edge deployment.
‼ production readiness: certificates - Warning
    Device is using self-signed, automatically generated certs.
    Please see https://aka.ms/iotedge-prod-checklist-certs for best practices.
√ production readiness: certificates expiry - OK
‼ production readiness: container engine - Warning
    Device is not using a production-supported container engine (moby-engine).
    Please see https://aka.ms/iotedge-prod-checklist-moby for details.
‼ production readiness: logs policy - Warning
    Container engine is not configured to rotate module logs which may cause it run out of disk space.
    Please see https://aka.ms/iotedge-prod-checklist-logs for best practices.
    You can ignore this warning if you are setting log policy per module in the Edge deployment.

Connectivity checks
-------------------
√ host can connect to and perform TLS handshake with IoT Hub AMQP port - OK
√ host can connect to and perform TLS handshake with IoT Hub HTTPS / WebSockets port - OK
√ host can connect to and perform TLS handshake with IoT Hub MQTT port - OK
√ container on the IoT Edge module network can connect to IoT Hub AMQP port - OK
√ container on the IoT Edge module network can connect to IoT Hub HTTPS / WebSockets port - OK
√ container on the IoT Edge module network can connect to IoT Hub MQTT port - OK
√ Edge Hub can bind to ports on host - OK

15 check(s) succeeded.
```
</details>

### Device (Host) Operating System
Windows 10 

### Architecture
amd64,

### Container Operating System
Linux containers

### Runtime Versions

#### iotedged
iotedge 1.0.8

#### Edge Agent
1.0.8

#### Edge Hub
1.0.8

#### Docker

```
Client: Docker Engine - Community
 Version:           19.03.2
 API version:       1.40
 Go version:        go1.12.8
 Git commit:        6a30dfc
 Built:             Thu Aug 29 05:26:49 2019
 OS/Arch:           windows/amd64
 Experimental:      false

Server:
 Engine:
  Version:          3.0.5
  API version:      1.40 (minimum version 1.24)
  Go version:       go1.12.1
  Git commit:       dbe4a30
  Built:            Thu Apr 18 22:07:58 2019
  OS/Arch:          windows/amd64
  Experimental:     false
```

## Logs

<details>
<summary>edge-agent logs</summary>

```
2019-09-24 07:45:18.506 +00:00 [WRN] - Reconcile failed because of the an exception
System.AggregateException: One or more errors occurred. (Error calling Create module publisher: Could not create module publisher
        caused by: Could not create module publisher
        caused by: **Container runtime error**
        caused by: expected value at line 1 column 1) ---> Microsoft.Azure.Devices.Edge.Agent.Edgelet.EdgeletCommunicationException: Error calling Create module publisher: Could not create module publisher
        caused by: Could not create module publisher
        caused by: Container runtime error
        caused by: expected value at line 1 column 1
```
</details>

marcschier, 2019-09-26:
Hi @ddobric.   Thanks for the details.  Could you also add the module deployment .json that was used?   Thanks a lot!

marcschier, 2019-09-26:
Also, is it possible to get the logs from the publisher, using iotedge logs <module>?  Assuming the publisher started but exited for any reason, crash, etc.   Thanks!

ddobric, 2019-09-27:
Here is deployment copied from portal. I also used 1.0.8 runtime. Result was the same.

Btw. There are no module logs, because module is not started.

```
{
    "modulesContent": {
        "$edgeAgent": {
            "properties.desired": {
                "modules": {
                    "publisher": {
                        "settings": {
                            "image": "mcr.microsoft.com/iotedge/opc-publisher:2.3",
                            "createOptions": "{\"Hostname\":\"publisher\",\"Cmd\":[\"--pf=/appdata/pn.json\",\"--aa\"],\"HostConfig\":{\"Binds\":[\"c:/iiotedge:/appdata\"]}}"
                        },
                        "type": "docker",
                        "status": "running",
                        "restartPolicy": "always",
                        "version": "1.0"
                    }
                },
                "runtime": {
                    "settings": {
                        "minDockerVersion": "v1.25",
                        "registryCredentials": {
                            "damir": {
                                "address": "damir.azurecr.io",
                                "password": "kkjOvM=J/sEyYTBW6TFltwuV5qXkVH70",
                                "username": "damir"
                            }
                        }
                    },
                    "type": "docker"
                },
                "schemaVersion": "1.0",
                "systemModules": {
                    "edgeAgent": {
                        "settings": {
                            "image": "mcr.microsoft.com/azureiotedge-agent:1.0",
                            "createOptions": ""
                        },
                        "type": "docker"
                    },
                    "edgeHub": {
                        "settings": {
                            "image": "mcr.microsoft.com/azureiotedge-hub:1.0",
                            "createOptions": "{\"HostConfig\":{\"PortBindings\":{\"8883/tcp\":[{\"HostPort\":\"8883\"}],\"443/tcp\":[{\"HostPort\":\"443\"}],\"5671/tcp\":[{\"HostPort\":\"5671\"}]}}}"
                        },
                        "type": "docker",
                        "status": "running",
                        "restartPolicy": "always"
                    }
                }
            }
        },
        "$edgeHub": {
            "properties.desired": {
                "routes": {
                    "route": "FROM /messages/modules/publisher INTO  $upstream"
                },
                "schemaVersion": "1.0",
                "storeAndForwardConfiguration": {
                    "timeToLiveSecs": 7200
                }
            }
        }
    }
}
```

marcschier, 2019-09-27:
Hi @ddobric - thanks so much.  Could you try removing the registryCredentials and setting the opc-publisher image to mcr.microsoft.com/iotedge/opc-publisher:latest?

ddobric, 2019-09-28:
Hi Marc,

I will try this. But usually if something with registry is not correct, edgAgent write out the error log message.

Damir

ddobric, 2019-09-28:
@marcschier  as expected, I get the same error after registry is removed.

> <4> 2019-09-28 16:03:04.286 +00:00 [WRN] - Reconcile failed because of the an exception
> System.AggregateException: One or more errors occurred. (Error calling Create module publisher: Could not create module publisher
>         caused by: Could not create module publisher
>         caused by: Container runtime error
>         caused by: expected value at line 1 column 1) ---> Microsoft.Azure.Devices.Edge.Agent.Edgelet.EdgeletCommunicationException: Error calling Create module publisher: Could not create module publisher

marcschier, 2019-09-28:
Thank you for trying, though.  We test on Windows 10 LTSB, so one last attempt I would like you to make is to a) on the machine run the container standalone using docker run command and b) try the latest release (tag is latest).  The former should give us more information regarding the container runtime error.  Can I also assume you are using Windows containers?  Or are you running under docker for Linux?

ddobric, 2019-10-01:
Thanks for feedback @marcschier.

I have founds that the issue is following:

`  "c:/iiotedge:/appdata"`

To me this looks correct. Unfortunately, when this is appended, container cannot start.
When using following container starts correctly, but it does not do what is supposed to do, because pn.json cannot be found.

```
 "Binds": [
      "c:/iiotedge:/appdata"
    ]
```

Damir

marcschier, 2019-10-02:
Thank you @ddobric, I will update the documentation to add the picture.  Will forward the second recommendation to the edge team.

marcschier, 2020-03-19:
Added the link to the issue and image to the docs.

closed 2020-03-19 by marcschier