Azure/Industrial-IoT #119: X509 4096-bit certificates as user credentials are too long

mm-dlwe, 2019-09-27:
**Describe the bug**
When trying to use 4096-bit RSA key the "Update endpoint information" in the opc registry service does not accept the base64 encoded PFX12 certificate because it is longer than 4096 bytes.

Error Message:
```
Error: Bad Request
Response body:
{
  "Message": {
    "Message": "ErrorCode:ArgumentInvalid;Property or Tag value should be maximum 4096 bytes. Error in Property/Tag <base64 encoded key> ",
    "ExceptionMessage": "Tracking ID:58f23afb8d89424082e59022a4850b7c-G:7-TimeStamp:09/27/2019 12:51:58"
  },
  "Data": {},
  "Source": "Microsoft.Azure.IIoT.Core",
  "HResult": -2146233088,
  "Exception": "BadRequestException"
}
```

**To Reproduce**
Steps to reproduce the behavior:
1. Generate a selfsigned PFX12 certificate with 4096-bit RSA Key
`openssl req -x509 -newkey rsa:4096 -keyout key.pem -out cert.pem -days 365 `
`openssl pkcs12 -export -out certificate.pfx -inkey key.pem -in cert.pem`
2. Encode the certificate using base64
3. Call Update endpoint information with the encoded certificate as value

**Expected behavior**
The certificate information should be accepted and strored for authenticating with the opcua server.

marcschier, 2019-09-29:
The problem is that the entire token is persisted.  We would need to make saving all credential types explicit.

marcschier, 2023-06-25:
Capability to use  certificates for user auth in OPC Publisher 2.9* will be tracked in #2005.

closed 2023-06-25 by marcschier