Azure/Industrial-IoT #120: Authentication information is removed from endpoint on scan

mm-dlwe, 2019-09-27:
**Describe the bug**
When submitting authentication data to an endpoint using "Update endpoint information" in registry service it is stored in the endpoint configuration. The problem is that this authentication data is removed as soon as a new scan for opc servers occurs.

**To Reproduce**
Steps to reproduce the behavior:
1. Discover an opc server
2. Add authentication data
3. Run another discovery
4. See that the authentication data is cleared again

**Expected behavior**
The authentication data should be not removed during scans.

marcschier, 2019-09-30:
Hi @mm-dlwe - after review we decided for security reasons to not persist any user secrets as part of the credential in the twin object.   We have thus determined to remove updating the endpoint information model with credential information.   Could you instead pass a credential into each individual call using the "header" object?   This should also take care of the X509Certificate size issue #119, as the payload in the header should not have any size restrictions (although this would add an overhead for each call of course).

If needed we could contemplate adding a mechanism to source credentials for a particular endpoint(id) from a separate authentication (micro)service using some form of service principal/token mechanism to our backlog for a future release.

mm-dlwe, 2019-10-11:
Howto pass the credential into each individual call using the "header" object?
Functions like the Activate endpoint method with starts the connection only have the endpointId parameter specified in swagger. There is no header object to specifiy a credential - at least i don't see it.
I can use the header.elevation entry in the twin api call but this does not seem to work if the endpoint is not connected. Activate endpoint seems to need the information as well but I don't see a possibility to provide it other than the update enpoint information call.
If that would work i think it would be acceptable to pass the auth data each time.

marcschier, 2019-10-13:
Ah - apologies, every request payload e.g. for browse, etc. has a header property which itself has a property called "Elevation" which can be used to pass username password credentials.

mm-dlwe, 2019-10-16:
Yes, i found that too but only for the twin service api calls. I should ask more precisely: How to pass the authentication data to the activate endpoint call in the registry service (when not using update endpoint information)?
I can specify only the endpoint id when i call it (no header). And if it doesn't have the authentication data the twin module will still try to connect immediately after the call and fail due to the missing authentication data.

marcschier, 2019-10-19:
Thanks so much, this explains it all, so the server cannot be connected to using anonymous.  Thanks, we need to think about this some more and get back.

monikavar, 2020-01-24:
The latest release does not have the ability to specify username/password in the entity itself anymore, instead you need to pass it with each call. Closing the issue.

closed 2020-01-24 by monikavar