Azure/Industrial-IoT #121: Error in twin service when using authentication

mm-dlwe, 2019-09-27:
**Describe the bug**
When using authentication data in combination authentication all operations return 401 Unauthorized. This happens with "UserName" as well as with "X509Certificate".
```
{
  "Result": 401,
  "Payload": "<valid jwt token>",
  "Message": "Response 401",
  "Data": {},
  "Source": "Microsoft.Azure.IIoT.Core",
  "HResult": -2146233088,
  "Exception": "MethodCallStatusException"
}
```

The twin module connects to the server and passes the auth data. the session is established. On executing a command the connection changes from Ready to Error:
```
[13:08:08 INF] Success - using publisher module 'publisher' (mcr.microsoft.com/iotedge/opc-publisher:latest) via methods.
[13:08:08 INF] Publisher connected!
[13:08:12 INF] 1: Endpoint opc.tcp://<Server>/ via opc.tcp://<Server>/ changed from Connecting to Ready
[13:09:01 INF] 1: Endpoint opc.tcp://<Server>/ via opc.tcp://<Server>/ changed from Ready to Error
[13:09:06 INF] 1: Endpoint opc.tcp://<Server>/ via opc.tcp://<Server>/ changed from Error to Ready
```
A wrong password shows a different error as expected:
```
[13:14:18 WRN] 2: Establishing unecrypted connection to opc.tcp://<Server>/.
[13:14:18 INF] 2: Creating session to opc.tcp://<Server>/ via opc.tcp://<Server>/.
[13:14:18 INF] New settings processed.
[13:14:18 INF] 2: BadUserAccessDenied creating session to opc.tcp://<Server>/ via opc.tcp://<Server>/ - retry.
[13:14:18 INF] 2: Endpoint opc.tcp://<Server>/ via opc.tcp://<Server>/ changed from Connecting to Error
[13:14:18 INF] 2: Try to connect to opc.tcp://<Server>/ via opc.tcp://<Server>/ in 1092 ms...
[13:14:19 WRN] 2: Establishing unecrypted connection to opc.tcp://<Server>/.
[13:14:19 INF] 2: Creating session to opc.tcp://<Server>/ via opc.tcp://<Server>/.
[13:14:19 INF] 2: BadUserAccessDenied creating session to opc.tcp://<Server>/ via opc.tcp://<Server>/ - retry.
[13:14:19 INF] 2: Try to connect to opc.tcp://<Server>/ via opc.tcp://<Server>/ in 3453 ms...
[13:14:23 WRN] 2: Establishing unecrypted connection to opc.tcp://<Server>/.
[13:14:23 INF] 2: Creating session to opc.tcp://<Server>/ via opc.tcp://<Server>.
[13:14:28 INF] 2: BadRequestTimeout creating session to opc.tcp://<Server>/ via opc.tcp://<Endpoint> - retry.
[13:14:28 INF] 2: Endpoint opc.tcp://<Server>/ via opc.tcp://<Server> changed from Error to NotReachable
[13:14:28 INF] 2: Try to connect to opc.tcp://<Server>/ via opc.tcp://<Server> in 300000 ms...
```

**To Reproduce**
Steps to reproduce the behavior:
1. Discover an opc ua server
2. Update the endpoint with authentication data
3. Activate the endpoint
4. Call for example "Browse" on the endpoint 
5. See error

marcschier, 2019-09-29:
Hi @mm-dlwe, can you provide more info around the server you are trying to access.  Also, an example request json with a sample user name password so I can build a test around it?  The access denied appears to point to an issue with how the credential is passed to the server.  Assuming the password is correct.

mm-dlwe, 2019-10-11:
Sure the server is the UACppServer Demo from Unified automation configured to accept authentication using X509 Certificates. I use this request for publishing a node in the twin server API:
```
{
  "item": {
    "nodeId": "ns=2;s=Demo.Dynamic.Arrays.Boolean",
    "publishingInterval": 10000,
    "samplingInterval": 10000
  },
  "header": {
    "elevation": {
      "type": "X509Certificate",
      "value": "MIIKsQIBAzCCCncGCSqGSIb3DQEHAaCCCmgEggpkMIIKYDCCBRcGCSqGSIb3DQEHBqCCBQgwggUE
AgEAMIIE/QYJKoZIhvcNAQcBMBwGCiqGSIb3DQEMAQYwDgQI2IhSzLuYluUCAggAgIIE0Krk8cEt
IVfu4RJCCGwZt5Ma37E9Jrp9edAK9zI0aVy0/SGEt+OvBVUICd9XcYixBIxZs7zeyhudDhCGdwJd
nN8/5rD8MK7MeFuvlor922hsYawJW6nEfIMrRkieIqw81X9w+9VwU70/3nPIES1nNNoXjbrEIgAt
yOjQ/xcxqRdedM3bx1v/Cye45Ao7JoMqhlVhyV2sf57Id4VzyYjcB/f+LcTCdX0r5f+wgxKYlL9S
NGvLiPactrZ41mhOInvCbmE0WHHqF3VRqWR2q/FI1m16odTmS3a3R0r0vfm/G9jstmIulV7sb7g5
Z4tKEBzcryUyJdqaaNPAE1oenkmltEFgkgj8i6mlz66mPfWnH8eEeC79+m7MT9AhkV7DiFh2Gzet
LWrLG5RxQkTjmUejG+tGfNKznF1bqn4B7bLavGyxdbUEIhb3ccudcVig78zKK8nXmIyHfzblR+ji
eS67TmaFVavxxu5SOHm8/TsIR/dn9dXRS0uVDzPvJuIqZUY/5FLPTkGV8tsZYUNY4+I28evQHLWi
ajk4S+hJJ3ah4DN8j3mLeRSpviBj0dOXG3CYpGBTMaBaIp+wIVrA+ThkMhmVzhvDZYP9D0f8VWRc
W15VGI+7JIBlP97633bSvkzxwUCbwtGVf7zook6WwEVm5z1rW6Nfqiac0EsMz0fSgaTUwRz5DVtH
NPBpiRf1X0uTH4eaXEmIsN33WKV3iHUSBORE4CSLBTXXobSFgkVPCLUWMKAeP6n3Wf1uhXH53UPo
W2f3e0LkmXZKy0NMiV223yc4ZMoTgyyeZudp6g039g3DWuENt+FSm+Ujo59LybtD2hkFQYTxUx89
c2S9GhwzE5sA+te78Nc9WW17vag8GkwenjatxIKVFd2IYjNXSCij+u/kR5/b1qbR57vpFvyPmb9S
PAZHVJT6JswQZ1EzicI6/l/J7+N0u0ZlxISFXR5HyOrnq6zcOFbnv/Aef5ZTfV39ZEcXcd98O2jt
xCZX397n6yUYSHyPjS6jUPiME6in7KmcYji0ajThCZ0szVg8tkRb+Mf1Eiwtir0d38vftn4oNnPQ
Lu73pJMiWNcQ2+FBLtWOaLi1C0risx0DZ/RIcZPQQ5tSiutXoBwapV0hUfcKlcKqsIugnaMTUd2B
hQgM3tFcU3NU47MgCMpk4eKW6816Snn/9NUWO9vFm3FOHNE0lvklEBbhO/FlnBV4CxuVM00oyYUm
n/6QbIE/x6G0Iq5Z1nPH5Fzes65I/EBNI38DDt3L/YpEYKtG33HjYVZjH4Tb/zMF/LdfYfyvhRFU
o2PnRGn2XfbHgKO5GhMHTMHWQceR4hh2/PwmAORK3/M7JMgYCCjPbfH3BHUHVqbehQ3UzqT9W3Uo
axCc1ANIJehASF+xxtsrPlhk3IFAm42Zs5gP0LWAE0OLw4UKHzww47Tn6ZZshfwRYNMurneLcovJ
L1Il6hUfOaM2v59mxoxPJNauZNGEvujQzTl3oUOA4pw9XPr5APl8tGpXzgJSLKyadaOVaYzkikCg
6g7Kg+UOZ3k+IKhv7Di7fFc4WHOsg0sExlAlPVvJOQubS8KnOmVJN5Fz/jRdEJhrEDR5rWB64hbv
A7JqJDXJrR8qhnuKU404wqBZ31QgftLnnXuUttlMMIIFQQYJKoZIhvcNAQcBoIIFMgSCBS4wggUq
MIIFJgYLKoZIhvcNAQwKAQKgggTuMIIE6jAcBgoqhkiG9w0BDAEDMA4ECBo9guXP+swsAgIIAASC
BMi8fxgwFapdJvb+pernQbVBdo+2k4dSczYfcZpVcJCYNW+j5+lDUu0Qt5dXLs0MJGTT2rwUND3I
xUZCZp4446I+rT+U7nWEZcCgQREgfVT1I3W8WB9IYVTnxe/oDEnQdo2fLSVpYXbLSgs+QUEnbyuO
bTXdggruz2/1V3s1qycYaRGzXyBj7WpyJ9jrWnBXvmfIlWmj5QEWYzbQBM5B+/b4S7Q9La2uoFF9
Sx8pFc7JyAw9QTT9E6bCH9YUA4S7PpdCeTxZ6gErCc3qnMlvhOc/j18RkGcR8+YKrJS6SgGPU61W
z9RuatQlSlH/2JQCMpbGf5XSs464gNHd5KQfNy45Y0u55bC4knXgt54e6fkk55nYjhYs91mSl6LA
juGEqpAGPWqGQlR5cz20fcIBKDqO20jFtwMh8nsB45OTsJHKSLniRo4an4kGUJHmQT1eCAAGnlve
vT4KXrgVHNc2z+PxeknS+PYkTccrVGdQClwvgpXPf066XGg1FQvZ+rkaNOLI4KgQ7KGAibPJEzwO
S3ILdk84g0kcjSZPnJKdQlNPdJ8akdPslyuGFYVNVmRJQkMmyzPCz0A7tenFnCwuIevTUegB141r
ZFfoFwGnlZC/XCvLYyR3kYWkFfj6KRd/2RGTtQ6Zd1mlnmPmQQlHJsAK8JKwlESXLaCNw6rTjWle
pQAprlsV6CzGBUB74lOO9K+GJKeOAGoyVvJLjHPs4xxC4yFS6B/jDX91Tyyhl4h4HcFcddCtM1xH
+PQL6bA9SP0ZgMbFqQcNQNnqAMrLRZjEhhiabw4ztE5dOGPCKzK07oiCFjQDwxAoCj8HAMHG6Jne
VgtXcpxP8/UzDP07dYwEYSNbFB2jT58k4pq4DAdVNhC5iNj7JMhXwYDUvCM0U+YNkCLHupPl4xog
ly3ETvOfZfJQKC8z2lAAr6JChqCCV7dY2FnoKCse8AdmnwX1/L/MbxHso0J0RofuXlFpWiFNDkS4
uVbGKoShCV+5FTaZj8ahj9kXbbakLvzg91U786XTPd51cHviAOuiLKTZMf3zYq/JmZlksvbJnFUw
SX5B4H/CyDDCgVoSianGz1Vh7twJeQxfzOHxIgiBZAfviUDCAdcEDBR6+NdzIKzGNN1qzdAB9SdK
ODwf7zdbR0R089rwq4K1FxK+6IQd8Unf1aAOoOKa3pZK7Bl8n7Efnxej9pEr1YOwW1anySrni4Qc
C2NDymZ7XUBXdLsb2Z6i+Mqiz2pC7TDCGOet6UN70+ojlqIcWGzg124hrFHOuA/3/v3E9D8hsJNh
XH0FMEEiGOcAnkS5JIvceDtJNRCVF4WjdAqGzdq6mtlFYj9yWJY7Hqv1e648SAxNs7Z7qOa36zJ9
W9m/Vw/Eyg/GstE1YHEnyVB+jAmu52eI1SL1FBZwcz7nr7F5wa0t4F22gTlRipY3udvRj9Vwd/zC
J1MUWPYjKCro40iNZK8VG6LmaPXK6JrFA7FmzbVSLdy+gkGcnOdwFHf2vbRdrLyH9oM4pUVdFJ2y
/y8Ba03l1Nvgjfxr+vYz5rlllRcXASrjQPU/Eo/G+8rDdbob5MneOYswDCPlxJ8DJPegr5UXOovy
A2NJbb08cREK/GOjkurx7qH01tjZqSYulyx89h8xJTAjBgkqhkiG9w0BCRUxFgQUe+AWiKoq/tlA
WH4tCabM6JL4iUwwMTAhMAkGBSsOAwIaBQAEFE6oMKZbxk6LATi0GsrYk0gN+W8WBAjGs6Xo9rGw
2wICCAA="
    }
  }
}
```

I put the test certificate in here as well so that yout can test with it (its a base64 encoded pfx). Wheter I use header.elevation or not does not matter. The authentication data was set using update enpoint information as well to get thje endpoint activation working.

The certificate is correctly whitelisted in the UaCPPServer and connections using UaExpert with the same certificate are working.

For user name password test we use the codesys opcua server of a WAGO PFC200. Which is a bit strange because it requires username and password to be send unencrypted.
But I would at least expect to get the UaCPPServer working.

marcschier, 2023-06-25:
Capability to use  certificates for user auth in OPC Publisher 2.9* will be tracked in #2005.

closed 2023-06-25 by marcschier