Azure/Industrial-IoT #123: OPC Publisher cannot connect to OPC Nodes because Cert issue

toddysm, 2019-09-30:
**Describe the bug**
We need to connect to legacy devices that support OPC-UA, collect the data they publish and send it to Azure IoT hub. For that purpose, we have the OPC Publisher deployed on a gateway device. However, because the legacy devices have self-signed certificates on them, the OPC publisher cannot connect to them and throws the following error
```
[19:09:56 INF] Connect and monitor session and nodes on endpoint 'opc.tcp://172.16.7.2:4840'.
[19:09:56 INF] Create unsecured session for endpoint URI 'opc.tcp://172.16.7.2:4840' with timeout of 50000 ms.
[19:09:56 ERR] Session creation to endpoint 'opc.tcp://172.16.7.2:4840' failed 21 time(s). Please verify if server is up and Publisher configuration is correct.
System.AggregateException: One or more errors occurred. (BadSecurityChecksFailed) ---> Opc.Ua.ServiceResultException: BadSecurityChecksFailed ---> Opc.Ua.ServiceResultException: BadSecurityChecksFailed
   at Opc.Ua.CertificateValidator.InternalValidate(X509Certificate2Collection certificates)
   at Opc.Ua.CertificateValidator.Validate(X509Certificate2Collection chain)
   --- End of inner exception stack trace ---
   at Opc.Ua.Client.Session.Create(ApplicationConfiguration configuration, ConfiguredEndpoint endpoint, Boolean updateBeforeConnect, Boolean checkDomain, String sessionName, UInt32 sessionTimeout, IUserIdentity identity, IList`1 preferredLocales)
   --- End of inner exception stack trace ---
   at System.Threading.Tasks.Task`1.GetResultCore(Boolean waitCompletionNotification)
   at System.Threading.Tasks.Task`1.get_Result()
   at OpcPublisher.OpcSession.ConnectSessionAsync(CancellationToken ct) in D:\a\1\s\modules\opc-publisher\opcpublisher\OpcSession.cs:line 444
---> (Inner Exception #0) Opc.Ua.ServiceResultException: BadSecurityChecksFailed ---> Opc.Ua.ServiceResultException: BadSecurityChecksFailed
   at Opc.Ua.CertificateValidator.InternalValidate(X509Certificate2Collection certificates)
   at Opc.Ua.CertificateValidator.Validate(X509Certificate2Collection chain)
   --- End of inner exception stack trace ---
   at Opc.Ua.Client.Session.Create(ApplicationConfiguration configuration, ConfiguredEndpoint endpoint, Boolean updateBeforeConnect, Boolean checkDomain, String sessionName, UInt32 sessionTimeout, IUserIdentity identity, IList`1 preferredLocales)<---
```
We assume the problem is that the legacy device uses an invalid certificate. We tried deploying the certificate as a trusted certificate in the gateway ca-repository but this didn't solve the issue.

Our assumption was that `unsecure` session means really not secure communication. What is the process for disabling the certificate validation?

marcschier, 2019-10-01:
Hi @toddysm, for quick proof of concept you can start the publisher with the "--aa" command line argument to accept all certificates (pass on command line or via CreateOptions).  Could you give this a try to see if this fixes your issue?

This flag is not recommended in production though and there is a slight issue with "expired" certificates which are in the current OPC UA stack version rejected before the stack hands them to the publisher validator callback.  

For production we currently recommend to map the PKI folders of the publisher client to a volume and then manually moving "rejected" certificates from the "rejected" folder into the "Trusted" folder.    This is similar to what you would likely have to do on the server side also.

We are planning to integrate both publisher and twin with OPC Vault so that the latter step will be cloud /remote manageable.

The "Unsecure" instructs the publisher to select an endpoint with "Security.None".  If the server does not have such endpoint or if it still presents a certificate during session establishment it is validated though.

toddysm, 2019-10-02:
@marcschier the `-aa` option works. We will play more with the `trusted` and `rejected` folders to see whether we can figure out how to manage this

toddysm, 2019-10-05:
Hi @marcschier I have to reopen this. It turns out that this worked in our test environment but not on the customer premises. We have exactly the same configuration in the test environment and on the customer premises but on the customer side, we see an error. OPC Publisher claims that the certificate has expired but it expires in 2022. Here the error from the logs:

```
[21:13:15 INF] Connect and monitor session and nodes on endpoint 'opc.tcp://172.16.7.3:4840'.
[21:13:15 INF] Connect and monitor session and nodes on endpoint 'opc.tcp://172.16.7.2:4840'.
[21:13:15 INF] Create unsecured session for endpoint URI 'opc.tcp://172.16.7.3:4840' with timeout of 50000 ms.
[21:13:15 INF] Create unsecured session for endpoint URI 'opc.tcp://172.16.7.2:4840' with timeout of 50000 ms.
[21:13:16 ERR] Session creation to endpoint 'opc.tcp://172.16.7.2:4840' failed 9 time(s). Please verify if server is up and Publisher configuration is correct.
System.AggregateException: One or more errors occurred. (Certificate has is expired or not yet valid. NotTimeValid: certificate has expired) ---> Opc.Ua.ServiceResultException: Certificate has is expired or not yet valid. NotTimeValid: certificate has expired ---> Opc.Ua.ServiceResultException: Certificate has is expired or not yet valid. NotTimeValid: certificate has expired
   at Opc.Ua.CertificateValidator.InternalValidate(X509Certificate2Collection certificates)
   at Opc.Ua.CertificateValidator.Validate(X509Certificate2Collection chain)
   --- End of inner exception stack trace ---
   at Opc.Ua.Client.Session.Create(ApplicationConfiguration configuration, ConfiguredEndpoint endpoint, Boolean updateBeforeConnect, Boolean checkDomain, String sessionName, UInt32 sessionTimeout, IUserIdentity identity, IList`1 preferredLocales)
   --- End of inner exception stack trace ---
   at System.Threading.Tasks.Task`1.GetResultCore(Boolean waitCompletionNotification)
   at System.Threading.Tasks.Task`1.get_Result()
   at OpcPublisher.OpcSession.ConnectSessionAsync(CancellationToken ct) in D:\a\1\s\modules\opc-publisher\opcpublisher\OpcSession.cs:line 444
---> (Inner Exception #0) Opc.Ua.ServiceResultException: Certificate has is expired or not yet valid. NotTimeValid: certificate has expired ---> Opc.Ua.ServiceResultException: Certificate has is expired or not yet valid. NotTimeValid: certificate has expired
   at Opc.Ua.CertificateValidator.InternalValidate(X509Certificate2Collection certificates)
   at Opc.Ua.CertificateValidator.Validate(X509Certificate2Collection chain)
   --- End of inner exception stack trace ---
   at Opc.Ua.Client.Session.Create(ApplicationConfiguration configuration, ConfiguredEndpoint endpoint, Boolean updateBeforeConnect, Boolean checkDomain, String sessionName, UInt32 sessionTimeout, IUserIdentity identity, IList`1 preferredLocales)<---
[21:13:16 ERR] Session creation to endpoint 'opc.tcp://172.16.7.3:4840' failed 9 time(s). Please verify if server is up and Publisher configuration is correct.
System.AggregateException: One or more errors occurred. (Certificate has is expired or not yet valid. NotTimeValid: certificate has expired) ---> Opc.Ua.ServiceResultException: Certificate has is expired or not yet valid. NotTimeValid: certificate has expired ---> Opc.Ua.ServiceResultException: Certificate has is expired or not yet valid. NotTimeValid: certificate has expired
   at Opc.Ua.CertificateValidator.InternalValidate(X509Certificate2Collection certificates)
   at Opc.Ua.CertificateValidator.Validate(X509Certificate2Collection chain)
   --- End of inner exception stack trace ---
   at Opc.Ua.Client.Session.Create(ApplicationConfiguration configuration, ConfiguredEndpoint endpoint, Boolean updateBeforeConnect, Boolean checkDomain, String sessionName, UInt32 sessionTimeout, IUserIdentity identity, IList`1 preferredLocales)
   --- End of inner exception stack trace ---
   at System.Threading.Tasks.Task`1.GetResultCore(Boolean waitCompletionNotification)
   at System.Threading.Tasks.Task`1.get_Result()
   at OpcPublisher.OpcSession.ConnectSessionAsync(CancellationToken ct) in D:\a\1\s\modules\opc-publisher\opcpublisher\OpcSession.cs:line 444
---> (Inner Exception #0) Opc.Ua.ServiceResultException: Certificate has is expired or not yet valid. NotTimeValid: certificate has expired ---> Opc.Ua.ServiceResultException: Certificate has is expired or not yet valid. NotTimeValid: certificate has expired
   at Opc.Ua.CertificateValidator.InternalValidate(X509Certificate2Collection certificates)
   at Opc.Ua.CertificateValidator.Validate(X509Certificate2Collection chain)
   --- End of inner exception stack trace ---
   at Opc.Ua.Client.Session.Create(ApplicationConfiguration configuration, ConfiguredEndpoint endpoint, Boolean updateBeforeConnect, Boolean checkDomain, String sessionName, UInt32 sessionTimeout, IUserIdentity identity, IList`1 preferredLocales)<---
```

Does the OPC publisher caches the information somehow? Should we just remove the module and recreate it? Or should we look for the problem somewhere else?

marcschier, 2019-10-06:
This could be related to the publishers certificate.  My first try would be to delete the publisher pki folder mapped as volume into the publisher container then restart.    How is the publisher started/running on prem?

toddysm, 2019-10-06:
It doesn't seem to be the publisher cert. Somewhere above in the logs, there are two messages saying that the node certs are not trusted and are saved in the rejected folder. Here are the container create options for the publisher:
```
{
  "Hostname": "opcpublisher",
  "ExposedPorts": {
    "62222/tcp": {}
  },
  "Cmd": [
    "publisher",
    "--ap=/appdata",
    "--pf=/appdata/pn.json",
    "--di=60",
    "--to",
    "--aa",
    "--si=0",
    "--ms=0",
    "--appcertstoretype=X509Store",
    "--ih=Amqp_Tcp_Only"
  ],
  "HostConfig": {
    "NetworkMode": "host",
    "PortBindings": {
      "62222/tcp": [
        {
          "HostPort": "62222"
        }
      ]
    },
    "CapAdd": [
      "NET_ADMIN"
    ],
    "Mounts": [
      {
        "Type": "bind",
        "Source": "/appdata",
        "Target": "/appdata"
      }
    ]
  },
  "NetworkingConfig": {
    "EndpointsConfig": {
      "host": {}
    }
  }
}
```

toddysm, 2019-10-10:
@marcschier We figured out the issue with the certs. It seems that the clocks were not synched on the industrial devices and this is the reason why the certs were not accepted by the publisher. Apologies for this - it seems the error was on our side.

closed 2019-10-10 by toddysm