Azure/Industrial-IoT #125: Some OPC UA server like Softing DataFEED OPC Suite cannot be scanned

ryan-ms, 2019-10-02:
**Describe the bug**
I've tried using the OPC Twin module to discover different OPC UA server in the same network including some software transfer from other protocols to OPC UA. Below is the test outcome

| OPC UA Server  | Test Status |
| ------------- | ------------- |
| Softing DataFEED OPC Suite running on windows   | Failed to discover  |
| OPC Foundation Sample Server runing on windows  | Failed to discover  |
| Prosys Simulation Server runing on windows  | Success to discover  |
| Siemens S7-1500   | Success to discover  |

**To Reproduce**
Steps to reproduce the behavior:
1. Running OPC Twin module within IoT Edge in Linux gateway (Ubuntu OS).
2. Use OPC Twin Web App to start scan process.
3. The issue happens: The OPC UA server from DataFEED OPC Suite cannot be found.

**Expected behavior**
All the OPC UA server can be auto discovered.

**Screenshots**

The DataFEED OPC Suite configuration
[image]

No result of DataFEED OPC UA server ( IP: 192.168.4.123) found in opc twin logs
[image]

**Additional context**
If you need more information, please let me know, thank you.

marcschier, 2019-10-04:
"Fast" scan only scans a limited set of ports, namely 4840/4841 and some in the high range.  4980 is not part of the range.   Either set port to 4840, or use "Scan" mode (can only be set through the CLI, not CF), i.e. "apps discover -d Scan" to do a full scan of all IANA unassigned ports on a machine.   

What was the endpoint address of the OPC sample server?  As that one's range should be part of the "Fast" port range list.

ryan-ms, 2019-10-08:
Hi @marcschier ,

Thank you, after I changed the OPC Server port to 4841. I can work now. I think the issue can be closed, thank you.

But there is another question. After change port,  I found that I need to enable ICMP-4 protocol in firewall setting, or the OPC server still cannot be scanned. Could you explain a little bit about this?

ryan-ms, 2020-02-14:
Hi @marcschier ,

> "Fast" scan only scans a limited set of ports, namely 4840/4841 and some in the high range. 4980 is not part of the range. Either set port to 4840, or use "Scan" mode (can only be set through the [...]

So you mentioned that 4840/4841 and some in the high range. Could you please help give the supported port list for OPC Twin fast scan?

Thank you.

monikavar, 2020-02-24:
Hi @ryan-ms , the supported port list for fast scan could be found here:
https://github.com/Azure/Industrial-IoT/blob/c070296e041ec6a0a42a7ff983a6cc1cd7c55fd1/common/src/Microsoft.Azure.IIoT.Net/src/Models/PortRange.cs#L139

Closing the issue as it is solved, please re-open in case of any other questions/issues.

closed 2020-02-24 by monikavar

ryan-ms, 2020-02-25:
Hi @monikavar ,
Thanks for your information. It really help.