Azure/Industrial-IoT #126: Publisher Base64 payload is not routable in IoT Hub

toddysm, 2019-10-03:
**Describe the bug**
We are using the OPC Publisher to collect information from devices and would like to route the information to different IoT Hub endpoints based on the payload we receive from the device. However, we noticed that the payload is Base64 encoded and the IoT Routing queries are not applied on properties in the body as described in the following article.

https://azure.microsoft.com/en-us/blog/iot-hub-message-routing-now-with-routing-on-message-body/

**To Reproduce**
Here is the message received from the OPC Publisher:
```
{
  "EnqueuedTimeUtc": "2019-10-03T17:58:46.6340000Z",
  "Properties": {},
  "SystemProperties": {
    "connectionDeviceId": "agitaretech-iot-test-gw",
    "connectionModuleId": "opc-publisher",
    "connectionAuthMethod": "{\"scope\":\"module\",\"type\":\"sas\",\"issuer\":\"iothub\",\"acceptingIpFilterRule\":null}",
    "connectionDeviceGenerationId": "637048513528304448",
    "contentType": "application/opcua+uajson",
    "contentEncoding": "UTF-8",
    "enqueuedTime": "2019-10-03T17:58:46.6340000Z"
  },
  "Body": "eyJOb2RlSWQiOiJucz0yO3M9VjAuNjMiLCJBcHBsaWNhdGlvblVyaSI6InVybjpldGgwOlJpdHRhbDpDTUNfSUlJX1VBX1NlcnZlciIsIkRpc3BsYXlOYW1lIjoibnM9MjtzPVYwLjYzIiwiVmFsdWUiOnsiVmFsdWUiOjIzLjcsIlNvdXJjZVRpbWVzdGFtcCI6IjIwMTktMTAtMDNUMTc6NTg6NDQuMDU1MDAwMCswMDowMCJ9fQ=="
}
```
The decoded body is:
```
{
  "NodeId": "ns=2;s=V0.63",
  "ApplicationUri": "urn:eth0:Rittal:CMC_III_UA_Server",
  "DisplayName": "ns=2;s=V0.63",
  "Value": {
    "Value": 23.7,
    "SourceTimestamp": "2019-10-03T17:58:44.0550000+00:00"
  }
}
```
The following messge routing query does not work in IoT Hub:
```
$body.NodeId = 'ns=2;s=V0.63'
```
Messages are not routed based on the condition above.

**Expected behavior**
Ability to route based on the $body properties.

marcschier, 2019-10-04:
This is related to the content type being set != application/json.   A quick fix could be to set it to application/json, but this a) could break other applications and b) will be broken in our next publisher release.

Since in future version of publisher as per customer ask we will be sending OPC UA binary (or OPC UA Json) and "Pub/Sub compliant" network messages from publisher, and both are not routable, we will anyway need an extra conversion module at edge side to convert to some simplistic, routable message format for this scenario.   

As such I am tracking this item as feature request to add such sample module, instead of closing.

marcschier, 2020-02-12:
We decided to close and track feature work separately.

closed 2020-02-12 by marcschier