Azure/Industrial-IoT #127: NodeId and DisplayName are the same in the OPC Publisher message

toddysm, 2019-10-03:
**Describe the bug**
Related to #126 the values for NodeId and DisplayName are the same in the OPC Publisher message. Using the UA Expert software, we are able to see that the NodeId and the DisplayName for the particular node are different. 
Also, the information provided as part of the OPC Publisher message loses the context of the Node, which makes it hard to determine what the measurement is about. 

**To Reproduce**
Here is a sample body from a message:
```
{
  "NodeId": "ns=2;s=V0.63",
  "ApplicationUri": "urn:eth0:Rittal:CMC_III_UA_Server",
  "DisplayName": "ns=2;s=V0.63",
  "Value": {
    "Value": 23.7,
    "SourceTimestamp": "2019-10-03T17:58:44.0550000+00:00"
  }
}
```

**Expected behavior**
The DisplayName is the friendly name of the node.
The full context of the node will be essential to determine the meaning of the measurement.

**Screenshots**
Here is a screenshot of the same node from UA Expert
[image]

toddysm, 2019-10-30:
Hi @marcschier - any update on this. It makes is really hard to determine what sensirs the data comes from.

markusstuhler, 2019-11-05:
Hi @toddysm,

can you try to run the publisher with the additional command line parameter "--fd true"? This will read the display name from the node.

Thanks!
Markus

toddysm, 2019-11-08:
@markusstuhler When this option is set to tru, we are getting the following error:
`[16:52:26 ERR] Unhandled OPC UA ServiceResultException 'BadObjectDeleted' when monitoring node 'ns=2;s=V0.75' on endpoint 'opc.tcp://192.168.7.190:4840'. Continue.` for all the monitored nodes

toddysm, 2019-11-08:
Hi @markusstuhler You can ignore the above. We don't know what changed in the last 2 weeks but changing the startup options for the container as you suggested above started throwing the errors mentioned and no data was sent to IoT Hub. Removing the option didn't solve the problem - now, we were not getting any data from the device. After some changes, and upgrading the firmware of the device we were able to again communicate with it.

closed 2019-11-08 by toddysm