Azure/Industrial-IoT #128: Bad continuation token passed during browsing.

marcschier, 2019-10-06:
There is an error that is thrown by Kepserver when the twin sends the request to browse achannel. The error thrown is associated with the continuation token being invalid. 

This originates from our micro services fand can be seen by using wire shark on the edge between the publisher and Kepserver. The investigation is to determine if the definition for the continuation token in OPC UA is being adhered to by both parties. The suspicion at this point is that our twin does not adhere to it because UA Expert can do the browsing.

chrisgclayton, 2019-10-17:
The micro service appears to be prepending "nsu" to the browse request (e.g. nsu=someEdge;ns=2;s=Channel1.Device1). The current question is whether this prepending is  compliance to the specification

marcschier, 2019-10-19:
Nsu is used to pass stable namespace uri as the index can change over time.  Nsu is used her because the uri is not a real uri.

closed 2019-10-19 by marcschier