Azure/Industrial-IoT #131: The latest OPC Twin module cannot connect to IoT Hub in some devices

ryan-ms, 2019-10-08:
**Describe the bug**
I used the latest OPC twin module of image URI as `mcr.microsoft.com/iotedge/opc-twin:latest` in IoT edge which running in Win10. But the connection with IoT Hub seems cannot be established.
Below is the log of OPC Twin.

` at DotNetty.Transport.Channels.Sockets.TcpSocketChannel.DoFinishConnect(SocketChannelAsyncOperation operation)
   at DotNetty.Transport.Channels.Sockets.AbstractSocketChannel.AbstractSocketUnsafe.FinishConnect(SocketChannelAsyncOperation operation)
   --- End of inner exception stack trace ---
   at DotNetty.Transport.Bootstrapping.Bootstrap.DoResolveAndConnectAsync(EndPoint remoteAddress, EndPoint localAddress)
   at Microsoft.Azure.Devices.Client.Transport.Mqtt.MqttTransportHandler.<>c__DisplayClass81_0.<<CreateChannelFactory>b__0>d.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at Microsoft.Azure.Devices.Client.Transport.Mqtt.MqttTransportHandler.OpenAsyncInternal(CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.Transport.Mqtt.MqttTransportHandler.OpenAsync(CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.Transport.ProtocolRoutingDelegatingHandler.OpenAsync(CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.Transport.ErrorDelegatingHandler.<>c__DisplayClass22_0.<<ExecuteWithErrorHandlingAsync>b__0>d.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at Microsoft.Azure.Devices.Client.Transport.ErrorDelegatingHandler.ExecuteWithErrorHandlingAsync[T](Func1 asyncOperation)
   --- End of inner exception stack trace ---
   at Microsoft.Azure.Devices.Client.Transport.ErrorDelegatingHandler.ExecuteWithErrorHandlingAsync[T](Func1 asyncOperation)
   at Microsoft.Azure.Devices.Client.Transport.RetryDelegatingHandler.<>c__DisplayClass32_0.<<OpenAsyncInternal>b__0>d.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at Microsoft.Azure.Devices.Client.Transport.RetryDelegatingHandler.EnsureOpenedAsync(CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.InternalClient.OpenAsync()
   at Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory.ModuleClientAdapter.CreateAsync(String product, IotHubConnectionStringBuilder cs, String deviceId, String moduleId, ITransportSettings transportSetting, TimeSpan timeout, IRetryPolicy retry, Action onConnectionLost, ILogger logger)
   at Microsoft.Azure.IIoT.Utils.Try.Options[T](Func1[] options)<---
[02:55:43 INF] 1: Session closed.
[02:55:43 INF] Application certificate with C87FC48BE01C6CAF8B3938B1916B126FA666B078, CN=Azure IIoT OPC Twin found in the certificate store
[02:55:43 INF] Setting new application certificate C87FC48BE01C6CAF8B3938B1916B126FA666B078, CN=Azure IIoT OPC Twin...
[02:55:43 INF] Adding own certificate to configured certificate store
[02:55:43 WRN] WARNING: Automatically accepting certificates. This is a security risk.
[02:55:43 INF] Running in iotedge production context.
[02:55:48 INF] Finding publisher to use as publish service...
[02:55:48 INF] Testing publisher module opcpublisher using methods...
[02:55:48 INF] 2: Session created.
[02:55:48 WRN] 2: Establishing unecrypted connection to opc.tcp://opcpublisher:62222/.
[02:55:48 INF] 2: Creating session to opc.tcp://opcpublisher:62222/ via opc.tcp://opcpublisher:62222/.
[02:55:48 WRN] Trusting Peer Certificate C6CEAA6AE044C867280DB38B5F0A72F362D7E361, CN=publisher due to AutoAccept(UntrustedCertificates) set!
[02:55:49 INF] Using publisher server on localhost.
[02:55:49 INF] 2: Endpoint opc.tcp://opcpublisher:62222/ via opc.tcp://opcpublisher:62222/ changed from Connecting to Ready
[02:55:49 INF] Publisher connected!
[02:56:04 INF] Module mc1121_opctwin connection status changed to Disconnected due to Retry_Expired.
[03:00:48 ERR] Error during module execution - restarting!
System.AggregateException: One or more errors occurred. (Transient network error occurred, please retry.) ---> Microsoft.Azure.Devices.Client.Exceptions.IotHubCommunicationException: Transient network error occurred, please retry. ---> DotNetty.Transport.Channels.ConnectException: LogError connecting to [fe80::d0f:e100:5fd3:8a0a%63]:8883 ---> System.Net.Sockets.SocketException: A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond
   at DotNetty.Transport.Channels.Sockets.SocketChannelAsyncOperation.Validate()
   at DotNetty.Transport.Channels.Sockets.TcpSocketChannel.DoFinishConnect(SocketChannelAsyncOperation operation)
   at DotNetty.Transport.Channels.Sockets.AbstractSocketChannel.AbstractSocketUnsafe.FinishConnect(SocketChannelAsyncOperation operation)
   --- End of inner exception stack trace ---
   at DotNetty.Transport.Bootstrapping.Bootstrap.DoResolveAndConnectAsync(EndPoint remoteAddress, EndPoint localAddress)
   at Microsoft.Azure.Devices.Client.Transport.Mqtt.MqttTransportHandler.<>c__DisplayClass81_0.<<CreateChannelFactory>b__0>d.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at Microsoft.Azure.Devices.Client.Transport.Mqtt.MqttTransportHandler.OpenAsyncInternal(CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.Transport.Mqtt.MqttTransportHandler.OpenAsync(CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.Transport.ProtocolRoutingDelegatingHandler.OpenAsync(CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.Transport.ErrorDelegatingHandler.<>c__DisplayClass22_0.<<ExecuteWithErrorHandlingAsync>b__0>d.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at Microsoft.Azure.Devices.Client.Transport.ErrorDelegatingHandler.ExecuteWithErrorHandlingAsync[T](Func1 asyncOperation)
   --- End of inner exception stack trace ---
   at Microsoft.Azure.Devices.Client.Transport.ErrorDelegatingHandler.ExecuteWithErrorHandlingAsync[T](Func1 asyncOperation)
   at Microsoft.Azure.Devices.Client.Transport.RetryDelegatingHandler.<>c__DisplayClass32_0.<<OpenAsyncInternal>b__0>d.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at Microsoft.Azure.Devices.Client.Transport.RetryDelegatingHandler.EnsureOpenedAsync(CancellationToken cancellationToken)
   at Microsoft.Azure.Devices.Client.InternalClient.OpenAsync()
   at Microsoft.Azure.IIoT.Module.Framework.Client.IoTSdkFactory.ModuleClientAdapter.CreateAsync(String product, IotHubConnectionStringBuilder cs, String deviceId, String moduleId, ITransportSettings transportSetting, TimeSpan timeout, IRetryPolicy retry, Action onConnectionLost, ILogger logger)
   at Microsoft.Azure.IIoT.Utils.Try.Options[T](Func1[] options)
   --- End of inner exception stack trace ---
   at Microsoft.Azure.IIoT.Module.Framework.Hosting.ModuleHost.StartAsync(String type, String siteId, String serviceInfo, IProcessControl reset)
   at Microsoft.Azure.IIoT.Modules.OpcUa.Twin.ModuleProcess.RunAsync() in D:\a\1\s\modules\opc-twin\src\ModuleProcess.cs:line 98
---> (Inner Exception #0) Microsoft.Azure.Devices.Client.Exceptions.IotHubCommunicationException: Transient network error occurred, please retry. ---> DotNetty.Transport.Channels.ConnectException: LogError connecting to [fe80::d0f:e100:5fd3:8a0a%63]:8883 ---> System.Net.Sockets.SocketException: A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond
   at DotNetty.Transport.Channels.Sockets.SocketChannelAsyncOperation.Validate()
   at DotNetty.Transport.Channels.Sockets.TcpSocketChannel.DoFinishConnect(SocketChannelAsyncOperation operation)
   at DotNetty.Transport.Channels.Sockets.AbstractSocketChannel.AbstractSocketUnsafe.FinishConnect(SocketChannelAsyncOperation operation)
[... 30 lines cut ...]
[03:00:54 INF] Using publisher server on localhost.
[03:00:54 INF] 3: Endpoint opc.tcp://opcpublisher:62222/ via opc.tcp://opcpublisher:62222/ changed from Connecting to Ready
[03:00:54 INF] Publisher connected!
[03:01:09 INF] Module mc1121_opctwin connection status changed to Disconnected due to Retry_Expired.
[03:02:42 INF] Application certificate with C87FC48BE01C6CAF8B3938B1916B126FA666B078, CN=Azure IIoT OPC Twin found in the certificate store
[... cut ...]
[03:02:53 INF] 1: Endpoint opc.tcp://opcpublisher:62222/ via opc.tcp://opcpublisher:62222/ changed from Connecting to Ready
[03:02:53 INF] Publisher connected!
[03:03:07 INF] Module mc1121_opctwin connection status changed to Disconnected due to Retry_Expired.`

It shows " connection status changed to Disconnected due to Retry_Expired." and no response anymore. 

But When I change the tag to `2.0.1-windows-amd64`. It works fine.

And In other device which running Ubuntu OS, the image URI as `mcr.microsoft.com/iotedge/opc-twin:latest` works fine, too.

ryan-ms, 2019-10-25:
Any updated?

marcschier, 2019-10-25:
Hi @ryan-ms, what version of edge are you using.

ryan-ms, 2019-11-04:
Hi @marcschier ,

Sorry for late respond. 
My edge version is 1.0.8.

mikkoh1, 2019-12-20:
Also had a similar issue. The Twin module is unreachable from the CLI or REST API. Data stream from the OPC UA server works (Publisher module) but the Twin module (latest) is invisible (supervisor) and not working. Both Publisher and Twin module are published from the Marketplace via IoT Hub and they are up and running. Also Edge runtime is set up accordingly.
In this case the OPC UA server is a real device and it has also a real PLC. They both run on the same subnetwork with the Edge device. 

According to logs Twin can't find Publisher module. How Twin module should be configured to give it a visibility?

M1880, 2019-12-20:
Hi,

Also had a similar issue. The Twin module is unreachable from the CLI or REST API. Data stream from the OPC UA server works (Publisher module) but the Twin module (latest) is invisible (supervisor) and not working. Both Publisher and Twin module are published from the Marketplace via IoT Hub and they are up and running. Also Edge runtime is set up accordingly.
In this case the OPC UA server is a real device and it has also a real PLC. They both run on the same subnetwork with the Edge device.

According to logs Twin can't find Publisher module. How Twin module should be configured to give it a visibility?

marcschier, 2020-01-23:
There was an issue with using AMQP - so we switched back to MQTT for now.    The new deployment deploys all modules into a virtual IoT Edge - we have not seen this issue anymore in this setup.   Please reopen if you see this outside of the simulated deployment/Edge installation in the wild.   In this case, please follow the troubleshooting steps in the documentation to obtain logs for both OPC Twin and the EdgeHub module, as well as the used createoptions (if different than the default).

closed 2020-01-23 by marcschier

ryan-ms, 2020-02-25:
Hi @marcschier ,

Thanks for your updated. It works for me.