Azure/Industrial-IoT #132: Not an issue but a Noob Question about multiple OPC Server with each has multiple nodes

ssvillamor23, 2019-10-09:
Hi, I am just new to Azure IOT. This is not an issue at all and its just a simple question. The question is: "whats the best way or approach to use the OPC Publisher with a scenario of just 1 iot edge device and has 10 PLC with each has multiple nodes?"

jmbrunskill, 2019-10-10:
Hi @ssvillamor23,
Do your PLC's have an OPC UA Server built in or are you planning to connect to them via a separate OPC UA Server?

If each of your PLCs has it's own OPC UA Server, you can setup the publishednodes.json file to have mulitple OPC UA endpoints, one for each PLC.

The file might look something like this:
```
   [
      {
        "EndpointUrl": "opc.tcp://plc1:62541",
        "UseSecurity": false,
        "OpcNodes": [
          {
            "Id": "i=2258",
            "OpcSamplingInterval": 2000,
            "OpcPublishingInterval": 5000,
            "DisplayName": "Current time"
          }
        ]
      },
      {
        "EndpointUrl": "opc.tcp://plc2:62541",
        "UseSecurity": false,
        "OpcNodes": [
          {
            "Id": "i=2259",
            "OpcSamplingInterval": 2000,
            "OpcPublishingInterval": 5000,
            "DisplayName": "Current time"
          }
        ]
      }
    ]
```

If your PLCs don't have a built in OPC UA server, you might need to set one up and connect to your PLCs from that server. The setup of this is highly dependent on your PLCs and OPC UA server type.

You might want to first test you OPC UA connection from a windows based OPC UA client before setting it up for collection using OPC Publisher.
Once such GUI tools ss UA Expert which can be downloaded for free here:
https://www.unified-automation.com/products/development-tools/uaexpert.html

ssvillamor23, 2019-10-17:
The PLC we have has its built in OPC UA Server. Thank for answering my question.  That would be a big help for me. I was wondering of how this OPC has been for many years.  Do they have a free OPC server or an opensource where I could use of create? I hope this OPC Publisher as well supports the old OPC when I implement it.

jmbrunskill, 2019-10-17:
@ssvillamor23 I think the OPC Publisher Tool is only planning to support OPC UA connections.
If your PLC has a builtin OPC UA server you should be able to connect though.

closed 2019-10-30 by marcschier