Azure/Industrial-IoT #135: OPC UA Heartbeats are sent relative to the SourceTimeStamp resulting in duplication of heartbeats on restarts.

jmbrunskill, 2019-10-11:
**Describe the bug**
When a heartbeat is setup for a node, it triggers the heartbeat relative to the SourceTimestamp received from the OPC UA server.
If the SourceTimestamp is in the past when the first update is received at the OPC Publisher, the heartbeats and generate values with a timestamp in the past. If there is more than one restart, this can result in duplicated timestamps being sent.

```
2019-10-10 09:15:26.756 [processor-thread-2] INFO  c.f.a.e.m.s.MessageProcessorService - Tracing 112d6b4-0e4-49905 received record:
 {nodeId='ns=2;s=112d6b4-0e4-49905', applicationUri='PTC.ThingWorxIC.1:UA%20Server',value={sourceTimestamp='2019-10-10T04:29:00.2917451Z', value=100}}
2019-10-10 09:21:26.843 [processor-thread-2] INFO  c.f.a.e.m.s.MessageProcessorService - Tracing 112d6b4-0e4-49905 received record:
 {nodeId='ns=2;s=112d6b4-0e4-49905', applicationUri='PTC.ThingWorxIC.1:UA%20Server',value={sourceTimestamp='2019-10-10T04:35:00.2917451+00:00', value=100}}
2019-10-10 09:27:27.586 [processor-thread-1] INFO  c.f.a.e.m.s.MessageProcessorService - Tracing 112d6b4-0e4-49905 received record:
 {nodeId='ns=2;s=112d6b4-0e4-49905', applicationUri='PTC.ThingWorxIC.1:UA%20Server',value={sourceTimestamp='2019-10-10T04:41:00.2917451+00:00', value=100}}
2019-10-10 09:33:26.862 [processor-thread-1] INFO  c.f.a.e.m.s.MessageProcessorService - Tracing 112d6b4-0e4-49905 received record:
 {nodeId='ns=2;s=112d6b4-0e4-49905', applicationUri='PTC.ThingWorxIC.1:UA%20Server',value={sourceTimestamp='2019-10-10T04:47:00.2917451+00:00', value=100}}

```
In these logs you can see there is a 4-5hr difference between the source timestamp and time the record is being sent.

**Issues with the current approach**

1. It can produce duplication in downstream systems such as Time Series Insights (The opc publisher service was restarted a few times resulting in duplicated timestamp collection)

[image]

2. It can appear to generate 'gaps' in data stream timestamps.
If the heartbeat is initialized with a historical timestamp at startup (eg. on a opc ua re-connection ) when the actual value changes it will appear like a gap between the last heartbeat time, and the new source timestamp.
[image]

3. Heartbeats appear like they are missing for recent data
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Setup an OPC UA Server with an unchanging node with an old Source Timestamp
2. Setup published nodes with heartbeat interval enabled
3. Observe the messages sent 

**Expected behavior**
I think heartbeats should probably be based on a server timestamp, rather than source timestamp or the timestamp should reflect the enqueue time for the heartbeat message.,

**System Info:**
* OS: Ubuntu 18.04
* OPCPublisher 2.5.1
* IoTEdge 1.0.8
* PTC.ThingWorxIC.1:UA%20Server

markusstuhler, 2019-11-19:
Hi @jmbrunskill
The code is (partially) broken. The timestamps should not be advanced at all. However, duplicated message content is expected as the Source- and Server-Timestamps are cached when the values didn't change. So the "gap" is merely the indication that the value didn't change within that time range. The real timestamp of the heartbeat message is indicated by the EnqueuedTime-Property of the IotHub message as it is read from the EventHub-endpoint.

We will offer a better heartbeat solution soon which will use OPC UA server capabilities to trigger heartbeat messages. We will leave this issue open to track this feature work.

MarekLani, 2019-12-23:
In the meantime, while new heartbeat mechanism is ready, how can I differentiate Heartbeat message from publish/notification message once processing messages from IoT Hub?

markusstuhler, 2020-02-20:
Hi @MarekLani,

currently, this is not possible as the heartbeat message basically is just a resent message from before. However, the new mechanism will be ready soon and we won't fix that in the current version. So I will close this issue for now.

Thanks!
Markus

closed 2020-02-20 by markusstuhler