Azure/Industrial-IoT #136: System.ArgumentOutOfRangeException: - Causes Queuing of messages to edgeHub

jmbrunskill, 2019-10-11:
**Describe the bug**
We are seeing queuing behavior with OPC Publisher that seems to correlate with this message in our opc publisher logs.
`System.ArgumentOutOfRangeException:   The specified timeout must represent a value between -1 and 2147483647,   inclusive.`

This error occurred on 3 different servers with in 1 hour of each other, so might be related to run time.

```
[13:08:02   INF] OpcPublisher status @ 10/10/2019 13:08:02 (started @ 09/30/2019   20:20:32)
--
[13:08:02   INF] ---------------------------------
[13:08:02   INF] OPC sessions (configured/connected): 1/1
[13:08:02   INF] OPC subscriptions (configured/connected): 1/1
[13:08:02   INF] OPC monitored items (configured/monitored/to remove): 59744/59744/0
[13:08:02   INF] ---------------------------------
[13:08:02   INF] monitored items queue bounded capacity: 250000
[13:08:02   INF] monitored items queue current items: 0
[13:08:02   INF] monitored item notifications enqueued: 222539990
[13:08:02   INF] monitored item notifications enqueue failure: 0
[13:08:02   INF] ---------------------------------
[13:08:02   INF] messages sent to IoTHub: 298515
[13:08:02   INF] last successful msg sent @: 10/10/2019 13:07:47
[13:08:02   INF] bytes sent to IoTHub: 78132680141
[13:08:02   INF] avg msg size: 261737
[13:08:02   INF] msg send failures: 0
[13:08:02   INF] messages too large to sent to IoTHub: 0
[13:08:02   INF] times we missed send interval: 0
[13:08:02   INF] number of events: 222539990
[13:08:02   INF] ---------------------------------
[13:08:02   INF] current working set in MB: 4896
[13:08:02   INF] --si setting: 10
[13:08:02   INF] --ms setting: 262144
[13:08:02   INF] --ih setting: Amqp_Tcp_Only
[13:08:02   INF]   ==========================================================================
[13:08:30   ERR] Error while processing monitored item messages.
System.ArgumentOutOfRangeException:   The specified timeout must represent a value between -1 and 2147483647,   inclusive.
Parameter   name: millisecondsTimeout
Actual   value was -2147483648.
at   System.Collections.Concurrent.BlockingCollection`1.ValidateMillisecondsTimeout(Int32   millisecondsTimeout)
at   OpcPublisher.HubCommunicationBase.MonitoredItemsProcessorAsync(CancellationToken   ct) in   D:\a\1\s\modules\opc-publisher\opcpublisher\HubCommunicationBase.cs:line 1643
[13:09:02   INF]   ==========================================================================
```

Immediately after this logging occurs, the monitored item queue starts increasing until the limit is hit.

```
[13:09:02   INF] OpcPublisher status @ 10/10/2019 13:09:02 (started @ 09/30/2019   20:20:32)
--
[13:09:02   INF] ---------------------------------
[13:09:02   INF] OPC sessions (configured/connected): 1/1
[13:09:02   INF] OPC subscriptions (configured/connected): 1/1
[13:09:02   INF] OPC monitored items (configured/monitored/to remove): 59744/59744/0
[13:09:02   INF] ---------------------------------
[13:09:02   INF] monitored items queue bounded capacity: 250000
[13:09:02   INF] monitored items queue current items: 12554
[13:09:02   INF] monitored item notifications enqueued: 222577434
[13:09:02   INF] monitored item notifications enqueue failure: 0
[13:09:02   INF] ---------------------------------
```

**To Reproduce**
Steps to reproduce the behavior:
Unsure of replication steps at this stage, appears to occur after a significant running period  possibly every ~ 7 days?

**Expected behavior**
Any exceptions should be handled gracefully and connections to edgeHub/IoT Hub should be maintained to avoid dataloss.

jmbrunskill, 2019-10-11:
FYI - I've tried to correlate the edgeHub logs to see if there's anything in those logs that could be related. 

Here's what I think might be relevant edgeHub logs (filtered out messages relating other modules)
```
<7> 2019-10-10 13:08:30.347 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.Amqp.LinkHandlers.EventsLinkHandler] - EventsLinkHandler processed 1 messages for nzs070iotedg01m/opcpublisher
<7> 2019-10-10 13:08:30.375 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.Core.Routing.RoutingEdgeHub] - Received 1 message(s) from nzs070iotedg01m/opcpublisher
<7> 2019-10-10 13:08:30.379 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.Amqp.LinkHandlers.EventsLinkHandler] - EventsLinkHandler processed 1 messages for nzs070iotedg01m/opcpublisher
<7> 2019-10-10 13:08:30.409 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.Core.Routing.RoutingEdgeHub] - Received 1 message(s) from nzs070iotedg01m/opcpublisher
<7> 2019-10-10 13:08:30.414 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.Amqp.LinkHandlers.EventsLinkHandler] - EventsLinkHandler processed 1 messages for nzs070iotedg01m/opcpublisher
<7> 2019-10-10 13:08:30.442 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.Core.Routing.RoutingEdgeHub] - Received 1 message(s) from nzs070iotedg01m/opcpublisher
<7> 2019-10-10 13:08:30.446 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.Amqp.LinkHandlers.EventsLinkHandler] - EventsLinkHandler processed 1 messages for nzs070iotedg01m/opcpublisher
<7> 2019-10-10 13:08:30.482 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.Core.Routing.RoutingEdgeHub] - Received 1 message(s) from nzs070iotedg01m/opcpublisher
<7> 2019-10-10 13:08:30.486 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.Amqp.LinkHandlers.EventsLinkHandler] - EventsLinkHandler processed 1 messages for nzs070iotedg01m/opcpublisher
<7> 2019-10-10 13:11:23.108 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.CloudProxy.ConnectivityAwareClient] - Received connection status changed callback with connection status Disconnected_Retrying and reason Communication_Error for nzs070iotedg01m/opcpublisher
<7> 2019-10-10 13:11:24.990 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.CloudProxy.ConnectivityAwareClient] - Received connection status changed callback with connection status Connected and reason Connection_Ok for nzs070iotedg01m/opcpublisher
<7> 2019-10-10 13:13:00.487 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.Core.IDeviceScopeIdentitiesCache] - Getting service identity for nzs070iotedg01m/opcpublishe
<7> 2019-10-10 13:13:00.487 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.Core.DeviceScopeAuthenticator] - Client nzs070iotedg01m/opcpublisher in device scope reauthenticated locally.
<7> 2019-10-10 13:13:00.487 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.Core.Authenticator] - Client nzs070iotedg01m/opcpublisher re-authenticated successfully
<7> 2019-10-10 13:13:00.487 +00:00 [DBG] [Microsoft.Azure.Devices.Edge.Hub.Core.ConnectionReauthenticator] - Reauthenticated client nzs070iotedg01m/opcpublisher successfully
```

To me It looks like the edgeHub Communication_Error occurs after the OPC Publisher Exception, so looks to me like it might be caused by an OPC Publisher issue rather than the other way around?

After the exception in OPC Publisher logs, no `Received 1 message(s) from nzs070iotedg01m/opcpublisher` events are shown in the edgeHub log.
Client reauthentication messages keep occuring successfully in edgeHub logs.

Restarting OPC Publisher resolves this issue (at least temporarily)

marcschier, 2019-10-13:
The code looks potentially problematic in regards to how the milliseconds are calculated, probably causing some overrun on the int when cast.   Investigating.

jmbrunskill, 2019-10-22:
@marcschier I've managed to compile that code and done some testing with this issue. I haven't been able to replicate it locally without artificially causing the overflow. My best guess is that some how the datetime could be shifting on the server?

However I'm thinking it might be appropriate to limit the millisecondsTillNextSend to the DefaultSendIntervalSeconds if configured?

That would avoid this issue, and at least degrade nicely in this scenario. Would you be interested in reviewing a pull request like this?

HubCommunicationsBase.cs : line ~1630
```
                            millisecondsTillNextSend = nextSendTime.Subtract(DateTime.UtcNow).TotalMilliseconds;
                            if (millisecondsTillNextSend < 0)
                            {
                                MissedSendIntervalCount++;
                                // do not wait if we missed the send interval
                                millisecondsTillNextSend = 0;
                            }
                            else if(millisecondsTillNextSend > TimeSpan.FromSeconds(DefaultSendIntervalSeconds).TotalMilliseconds)
                            {
                                Logger.Error($"Invalid wait time till next send ({millisecondsTillNextSend})ms, Next Send Time: {nextSendTime}, Now {DateTime.UtcNow}].");
                                //Use Default Send interval seconds as timeout, If millisecondsTillNextSend is > Int32.MaxValue then the TryTake below will fail with an exception
                                millisecondsTillNextSend = TimeSpan.FromSeconds(DefaultSendIntervalSeconds).TotalMilliseconds;
                            }
```
I'm also concerned that other exceptions could cause a similar problems. Is there a nice way to restart the MonitoredItemsProcessorAsync Task if it fails for some reason?

marcschier, 2019-10-23:
Hi @jmbrunskill, the overall compare of double against 0 can lead to unexpected results.  It is better to first cast to an integer (probably long) and then check the absolute result, otherwise, this is probably the way to go.   I would probably rather catch the "failure" early on in some way and handle it gracefully.

markusstuhler, 2019-12-03:
Hi,

we included this fix in 2.5.2 so I'm closing this issue. Please reopen if the problem persists.

Thanks!
Markus

closed 2019-12-03 by markusstuhler