Azure/Industrial-IoT #138: OpcPublisher: BadSubscriptionIdInvalid after a while

dcrupel, 2019-10-21:
Hi,

We have an OpcPublisher running. It's running fine for days and suddenly it stops sending data for the subscription with the lowest publishing interval (1000ms). This is happening on two different Edge devices, 1 connected to a Siemens OPC UA server and 1 connected to a Kepware server.

**To Reproduce**
Steps to reproduce the behavior:
1. Install IoT Egde 1.0.9-rc1 on Windows 10 agent PC
2. Push node configuration to the device with 3 different publishing intervals (1000ms, 5000ms and 30000ms) = 3 subscriptions
3. Wait for 1,2,3,... days
4. See error (iotedge logs OpcPublisher --tail 100 -f) --> Subscription with lowest interval loses connection

**Expected behavior**
Publisher keeps subscriptions.

**Log**
{"log":"[07:28:53 ERR] Exception\r\n","stream":"stdout","time":"2019-10-17T07:28:53.8924897Z"}
{"log":"Opc.Ua.ServiceResultException: BadSubscriptionIdInvalid\r\n","stream":"stdout","time":"2019-10-17T07:28:53.8924897Z"}
{"log":"   at Opc.Ua.Client.Subscription.SetPublishingMode(Boolean enabled)\r\n","stream":"stdout","time":"2019-10-17T07:28:53.8924897Z"}
{"log":"   at OpcPublisher.OpcSession.MonitorNodesAsync(CancellationToken ct) in C:\\app\\opcpublisher\\OpcSession.cs:line 407\r\n","stream":"stdout","time":"2019-10-17T07:28:53.8924897Z"}
{"log":"[07:29:03 ERR] Exception\r\n","stream":"stdout","time":"2019-10-17T07:29:03.8999995Z"}
{"log":"Opc.Ua.ServiceResultException: BadSubscriptionIdInvalid\r\n","stream":"stdout","time":"2019-10-17T07:29:03.8999995Z"}
{"log":"   at Opc.Ua.Client.Subscription.SetPublishingMode(Boolean enabled)\r\n","stream":"stdout","time":"2019-10-17T07:29:03.8999995Z"}
{"log":"   at OpcPublisher.OpcSession.MonitorNodesAsync(CancellationToken ct) in C:\\app\\opcpublisher\\OpcSession.cs:line 407\r\n","stream":"stdout","time":"2019-10-17T07:29:03.8999995Z"}
{"log":"[07:29:13 ERR] Exception\r\n","stream":"stdout","time":"2019-10-17T07:29:13.903782Z"}
{"log":"Opc.Ua.ServiceResultException: BadSubscriptionIdInvalid\r\n","stream":"stdout","time":"2019-10-17T07:29:13.903782Z"}
{"log":"   at Opc.Ua.Client.Subscription.SetPublishingMode(Boolean enabled)\r\n","stream":"stdout","time":"2019-10-17T07:29:13.903782Z"}
{"log":"   at OpcPublisher.OpcSession.MonitorNodesAsync(CancellationToken ct) in C:\\app\\opcpublisher\\OpcSession.cs:line 407\r\n","stream":"stdout","time":"2019-10-17T07:29:13.903782Z"}

**Desktop (please complete the following information):**
 - OS: Windows 10 LTSC
 - edgeHub and edgeAgent = 1.0.9-rc1

**Additional context**
We now created 2 seperate edge devices which are monitoring the same OPC server, that way we can find out or the 2 modules loses the subcription at the same time (OPC UA server is causing the issue) or one module loses the subcription and the other doesn't (OpcPublisher module (or edge) is causing the issue)

mregen, 2019-10-23:
Hi @dcanfyn, double checking: With lowest interval you mean the 30s subscription?

pleinolijf, 2019-10-23:
> 
> 
> Hi @dcanfyn, double checking: With lowest interval you mean the 30s subscription?

No, 1000ms.

marcschier, 2019-10-23:
Apologies, but one more ask - I checked back with the team, and we recently fixed an issue that prevented a "correct" session "reconnect" in some cases.  The version with the fix is 2.5.1.   Is the issue happening on 2.5.1, or would it be possible to update the publisher to 2.5.1 and regress?

dcanfyn, 2019-10-24:
> Apologies, but one more ask - I checked back with the team, and we recently fixed an issue that prevented a "correct" session "reconnect" in some cases. The version with the fix is 2.5.1. Is the issue happening on 2.5.1, or would it be possible to update the publisher to 2.5.1 and regress?

We did change it now and are testing if this solves the problem. Thanks for the response. I'll close this issue if it's fixed.

pleinolijf, 2019-12-10:
> We did change it now and are testing if this solves the problem. Thanks for the response. I'll close this issue if it's fixed.

v2.5.1 does not fix the problem for us. We are collaborating with Microsoft on this.

pleinolijf, 2019-12-30:
v2.5.2 incorporates code that will automatically renew the subscription upon catching the exception. While this does not fix the cause of the error, it does minimize to the best extent any data loss, so this is an acceptable solution for us and the client.

pleinolijf, 2020-02-10:
Some extra feedback for anyone stumbling upon this issue: v2.5.2 contains another bug causing the module to stop sending any new messages after the keepalive request fails.

This was fixed in **v2.5.3** (not yet officially released at time of writing).  After some weeks on acceptance, we are running that now in production.
Issue may be closed, as far as I am concerned.

omarine, 2020-02-14:
Hello,

do you have an official date for the release of the v2.5.3 ?
[image]

guybartal, 2020-02-24:
Hi,

I need this fix as well for my customer, can I we have clarity when will it be released?

Meanwhile is there a stable branch with that fix that we can use for a POC?

chmagitt, 2020-03-05:
Hello 

Can you please share the date for 2.5.3 ?

marcschier, 2020-03-10:
2.5.3 fixes this, and has been released several moons ago but we missed tagging it - I just uploaded the tag - you can find the codebase in the release/2.5.3 branch and with "2.5.3" and "2.5" tag on mcr.  We are going to fix the MCR presence soon, so it should show up on dockerhub as well.   This is also been fixed in 2.6* (aka master) even though the code there is completely new.

closed 2020-03-10 by marcschier