Azure/Industrial-IoT #139: RuntimeBinderException in Microsoft.Azure.IIoT.Hub.Client.IoTHubServiceClient.GetAsync

mma-mm, 2019-10-23:
Hi,

I'm trying to retrieve a module twin using the IoTHubServiceClient via the IIoTHubTwinServices. Object was created by Autofac. I'm using version 2.5.1 of the corresponding NuGet. This leads to the following exception:

    Exception has occurred: CLR/Microsoft.CSharp.RuntimeBinder.RuntimeBinderException
    Exception thrown: 'Microsoft.CSharp.RuntimeBinder.RuntimeBinderException' in 
        System.Private.CoreLib.dll: 'Cannot convert type 'Microsoft.Azure.Devices.Shared.TwinCollection' to 'Newtonsoft.Json.Linq.JToken''
        at Microsoft.Azure.IIoT.Hub.Client.IoTHubServiceClient.<GetAsync>d__9.MoveNext()

I'm confused by this kind of exception, as TwinCollection seems to be convertible to JToken.
 
Is there anything i need to take into account when using the IoTHubServiceClient?

marcschier, 2019-10-24:
Hi @mma-mm, We also just found this, and have made a fix, which is not in master yet.  Using this to track.

marcschier, 2020-02-28:
With the current release, we have decided to not ship any non-API nuget packages any longer.   Hence will close as wont fix.

closed 2020-02-28 by marcschier